Ultimaker/Cura #14790: Tool tips fade immediately when the mouse is over special icons in the print settings dropdown.

dmarnel, 2023-03-06:
### Application Version

5.2.2

### Platform

macOS 13.2.1 (22D68)

### Printer

Ender 3 Pro

### Reproduction steps

Click the Print Settings dropdown and move the mouse over one of the special icons next to a setting, such as the **counter-clockwise arrow**, the **fx** or the **link** icon.

### Actual results

 The tool tip appears only for a fraction of a second and then disappears.

### Expected results

The tool tip should remain visible until the mouse is moved away.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No addition info.

MariMakes, 2023-03-07:
Hey @dmarnel,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This is what I am seeing:
[image]

Is your issue that the tooltips disappear when you have a clickable arrow, or Fx symbol? 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

dmarnel, 2023-03-07:
> Is your issue that the tooltips disappear when you have a clickable arrow, or Fx symbol?

Thanks, @MariMakes, yes, and I believe that is a fairly new bug. It was working not too long ago as I recall. If you move the mouse back and forth over one of those special symbols you can see the explanation tool tip appear but then it immediately disappears.

commit aec8a9ad (2023-03-08): Fix the hover for SettingItem

This was broken due to the switch to Qt6

Fixes CURA-10375 and fixes #14790
closed 2023-03-08 by nallath