Ultimaker/Cura #14793: Cura 5.2.1-5.3.0-beta-2 doesn't open

etkaar, 2023-03-06:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

FLSUN Q5

### Reproduction steps

I tried it with version 5.2.1-5.3.0-beta-2, but no success. Cura will not open and instead show a white page in the preview:

[image]

I use two monitors. They are the same model and resolution (4K). Instead of loading on the primary monitor, Cura shows the loading screen on the secondary monitor:

[image]

### Actual results

Not opening

### Expected results

Open

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10900906/cura.log)

MariMakes, 2023-03-07:
Hey @etkaar,

Sorry to hear you are stuck 😞 

I have a couple of questions to narrow it down.
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have two monitors placed above each other? Can you try this workaround from the community? https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105
	3. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	4. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	5. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

MariMakes, 2023-03-31:
Quick update on our side,

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download link and discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)

etkaar, 2023-03-31:
Thank you! The issue suddenly disappeared, which is good and bad the same time. If it occurs again, I will ty that!

MariMakes, 2023-04-03:
Thanks for getting back to us.
I'll close the issue 🙂 

Happy printing!

closed 2023-04-03 by MariMakes