Ultimaker/Cura #14794: (5.2.2) Slicing failed with an unexpected error

TripleWhy, 2023-03-06:
### Application Version

5.2.2

### Platform

Windows 11

### Printer

Anycubic Vyper

### Reproduction steps

1. Open the attached project: [AV_karpador.svg_0.6mm_v3.zip](https://github.com/Ultimaker/Cura/files/10902489/AV_karpador.svg_0.6mm_v3.zip)
2. Click slice

Notes: It sliced fine before, but after changing some print settings, I got this message.

### Actual results

Slicing failed

Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.
[image]

### Expected results

Successful slicing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10902531/cura.log)

MariMakes, 2023-03-08:
Hey @TripleWhy,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I can reproduce your slicing error on my side, so I'll bring it up to the team.

I was able to slice your file in Cura 5.3 after rotating it 90 degrees, so it probably is related to rounding. I hope that helps you get unstuck. 
[image]

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-1S](https://ultimaker-o7.sentry.io/issues/4859789659/)