Ultimaker/Cura #14795: Changes to "machine settings" do not "stick"

Loki44, 2023-03-06:
### Application Version

5.3.0-beta.2

### Platform

windows 10

### Printer

Ultimaker 2+

### Reproduction steps

1. Install cura
2. add machine ultimaker 2+
3. go to manage printers -> machine settings
4. change "g-code flavor" to "marlin"
5. close the machine settings **and** preferences window
6. go to manage printers -> machine settings
7. see that "g-code flavor" is still "ultimaker 2"

### Actual results

"g-code flavour" is **not** changed permanently 

### Expected results

"g-code flavour" is changed permanently 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I'm using marlin 2 on my ultimaker 2+ because of issues I was having with the stock firmware. 

I've been able to change this just fine in the past, so it's a relatively new issue.  

There's no project file, because it happens with a blank buildplate.

log below.

[cura.log](https://github.com/Ultimaker/Cura/files/10902770/cura.log)

GregValiant, 2023-03-06:
Thanks for the report.
I installed a UM printer in Cura and the problem showed up.  After leaving the dialog and re-opening it the gcode flavor is reset to Griffon.
Someone from the UM team will take a look.

MariMakes, 2023-03-07:
Hey @Loki44,

Thanks for your report 👍
I can reproduce it, it seems to be introduced in the 5.3 Beta.

I'll bring it up to the team to see what they can do to improve it.
Fingers crossed 🤞

Loki44, 2023-03-13:
Thanks.

I don't know if this helps locate the bug, but on an installation where the setting was changed in 5.2, then upgraded, the setting is kept.

samori81, 2023-03-20:
Also having this flavor saving problem. We are using 3D printer OS and need Marlin flavor to operate our 30x UM2+ University  3D printing farm. I hope the issue is fixed quickly.

GregValiant, 2024-06-22:
This bug has been addressed in current versions of Cura. I'll go ahead and close this.

closed 2024-06-22 by GregValiant