Ultimaker/Cura #14802: Pause at Height

ShadMan5, 2023-03-07:
### Application Version

5.3.2

### Platform

Windows 11

### Printer

Ultimaker S5 pro bundle

### Reproduction steps

Print job is programed with 2 paus instances at particular layer numbers
Printer properly pauses and lowers at the first instance
Printer resumes printing at leased 10mm lower than it should after resumption of printing.

This problem manifests on 2 different printers with same setup.
Printer firmware used on both printers is 8.1.1

### Actual results

Printer resumes after paus >10mm lower than it should (part is >10mm below the nozzle on restart)

### Expected results

Printer resumes printing where it left off on the part being printed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

this problem did not manifest with the older version of Cura and printer firmware.

GregValiant, 2023-03-07:
Thanks for the report.
I am unable to reproduce this.  We will need one of those "bad" gcode files and a project file to investigate.  Load a model, set Cura up to slice with the pauses, and use the "File | Save Project" command to create a 3mf project file.  Put the project file and the gcode into a zip folder and post the zip folder here.

Pause At Height moves the Z up 1mm from the last Z before the pause and then drops it back down that 1mm when the pause resumes.  At least that's how it is supposed to work (and does when I do it).  Here is a snippet of what I get when I use it:
**G0 F600 X128.607 Y125.837 Z15.3**  >>> last Z before the pause
G0 F9000 X121.781 Y108.412
G0 X117.824 Y106.789
;TIME_ELAPSED:1540.752496
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 50
M83 ; switch to relative E values for any needed retraction
**G1 F300 Z16.3 ; move up a millimeter to get out of the way**
G1 F9000 X0 Y0
M104 S205 ; standby temperature
M84 S3600 ; Keep steppers engaged for 1h
M300 S440 P1000 ; Beep
M0 ; Do the actual pause
M104 S240 ; resume temperature
G1 F9000 X117.824 Y106.789
**G1 F300 Z15.3 ; move back down to resume height**
G1 F2100 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E2737.55882

ShadMan5, 2023-03-07:
Hello,
I just updated Cura to release 5.3.0 and still same issue.  I use Fusion 360 to produce the 3mf files as I have for many months.

Right now:

  1.  The machine pauses
  2.  Bed lowers all the way down
  3.  I insert the nuts in their proper slots
  4.  When I press resume, the machine raises bac up to within 15mm of the nozzle and the process continues.  The result is extruded plastic in that 15mm gap which results in spaghetti strands.

Please advise as soon as possible.  I have a time sensitive project and would appreciate your help in resolving this matter.

All the best,
-Farshad

***@***.***

Farshad Bagheri
BagheriTechnologies, Inc.
7001 Central Ave, Suite 1
Saint Petersburg, FL  33710
Tel: (727) 204-7633 | Fax: (727) 341-0121

Valentin-Wilhelm, 2023-03-09:
Hi,

i have a simmilar issue. I want to change the printbed after leveling. I added the following as the start gcode in cura:
```
M117 change Printbed
M25
G0 Z200
G92 Z200.8
```
There is a G92 because the new Printbed is 0.8 mm less high as the one for leveling. (the new one cant be used for leveling because the hot nozzle would melt it)

This setting results in the following correct code in the file:
```
[...]
;Generated with Cura_SteamEngine 5.2.1
T0
M82 ;absolute extrusion mode
M117 change Printbed
M25
G0 Z200
G92 Z200.8
G92 E0
M109 S240
G280 S1
G0 Z20.001
G1 F2700 E-8
;LAYER_COUNT:66
;LAYER:0
[...]
```
However when I run the code the nozzle is about 10 to 15 mm above the Printbed, simmilar the your Problem. This setting worked for older firmware versions, sadly I cant remember version. I alredy tried to change the order of the the comands, Canged G0 to G1 which did not solve the problem. I also removed the M25 and swaped the plate on the fly, which also resulted in the unwanted offset. I checked if the possition was correct by G0 Z0.2 before G92 which resulted in a correct possition of 0.2 mm above the bed.

I suspect there is something wrong with the firmware of the printer.

Cura 5.2.1
Ultimaker S5 Firmware 7.0.3

Gcode with TPU on extruder 1:
[UMS5_walze.zip](https://github.com/Ultimaker/Cura/files/10930257/UMS5_walze.zip)