Ultimaker/Cura #14806: [PostProcessingPlugin] Ability to use M125 command with PauseAtHeight.py

Simonlebucheron, 2023-03-07:
### Is your feature request related to a problem?

In the PauseAtHeight post process script we are not able to choose the [`M125`](https://marlinfw.org/docs/gcode/M125.html) pause method, to let the machine firmware pause and park the head at the defined height/layer.

### Describe the solution you'd like

Be able to choose `M125` in the method field of the Pause at height post-process script windows.

### Describe alternatives you've considered

- `M0` and `M25` are not the right dedicated function.
- `@pause` with octoprint should work only if we have configured the _[afterPrintPaused script](https://docs.octoprint.org/en/master/features/gcode_scripts.html?highlight=pause#id4)_ into octoprint

### Affected users and/or printers

- Marlin firmware machines
- Cura and PostProcessingPlugin

Simonlebucheron, 2023-03-08:
> Have you tried putting M125 into the "gcode before pause" or "gcode after pause" box?

@GregValiant  , good point, I have assumed there will be conflict between machine firmware pause management and octoprint pause management, I ll test it and inform you.

GregValiant, 2023-03-08:
Yes, there might be a conflict because there will be either M25 or M0 in the gcode in addition to the M125.  If either M25 or M0 does not work on your printer, that would be the one to use.
The Search and Replace plugin would also be an option.  It would need to run AFTER Pause at Height.
If you were to use the M0 command in Pause at Height then:
Search: M0
Replace: M125
Regular Expressions = False

The alteration of Pause at Height to include M125 is an addition of only two lines to the script.

Simonlebucheron, 2023-03-08:
> Yes, there might be a conflict because there will be either M25 or M0 in the gcode in addition to the M125. If either M25 or M0 does not work on your printer, that would be the one to use. The [...]

I prefer this method, I think that is the good way to go for now. Is there a reason why `M125` as pause command is not implemented yet ? I think it will be use more in the future.

GregValiant, 2023-03-08:
It depends on the firmware implementation and it appears to be an OctoPrint thing.  The RepRap Wiki indicates that M125 is not enabled in any firmware flavors.  It is certainly not enabled in the Marlin version on my printer.
Looking deeper into Pause At Height it certainly isn't as simple as I first made out.  The X, Y, Z and P parameters may have to be passed with M125 and disabled in the gcode as there may be a conflict.  Without a printer that it works on I can't check.

Simonlebucheron, 2023-03-15:
Ok, meanwhile this is implemented in PostProcessingPlugin, Octoprint team help me figured out [the way to go](https://github.com/OctoPrint/OctoPrint/issues/4751) 

To correctly pause a print for a marlin machine with Octoprint and Cura:
1. In Octoprint configure `M125` as a pause trigger (Settings > Serial Connection > Firmware & Protocol > Protocol Fine Tuning > Pausing commands)
2. In Octoprint write `M125' into your pausing code in Settings > GCODE Scripts > After print job is paused.
3. In Cura, with PostProcessingPlugin and PauseAtHeight script use `Repetier/Octoprint (@pause)` method and untick `Park Print`

Note: I have set `M0, M1, M25, M125` as a pause trigger in octoprint (as safety net) so we can use any related method with PauseAtHeight or any custom Gcode.