Ultimaker/Cura #14810: NO Support Brim anymore if 'Built Plate Adhesion' is set to 'None'

discip, 2023-03-07:
### Application Version

5.3.0

### Platform

Manjaro

### Printer

unrelated

### Reproduction steps

1. In **`Support`** tick the box for `Enable Support Brim`.
2. In **`Built Plate Adhesion`** set `Built Plate Adhesion Type` to _`None`_

### Actual results

The entire initial support - layer is missing.

### Expected results

The first layer for supports should print even if `Built Plate Adhesion Type` is set to _`None`_ as before!

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

# 5.2.2
[image]

# 5.3.0
[image]

GregValiant, 2023-03-08:
Thanks for the report.
I have seen this myself and I recall another bug report on this behavior (#14725).
I'll mark this as a duplicate report and the team will take a look at this.

discip, 2023-03-09:
@GregValiant
Should I close this issue as it is a duplicate? (Sorry, I did't search before posting.)

discip, 2023-03-09:
@GregValiant
Should I close this issue as it is a duplicate? (Sorry, I did't search before posting. 😓)

GregValiant, 2023-03-10:
Let's leave it open.  Sometimes the Cura team needs more examples of a problem in order to properly research it.

RaptorTheOne, 2023-04-08:
I want to report that this issue also affects the plugins "Tab Anti Warping" and "Tab+ AntiWarping".
I spent 8 hours playing around with finding a workaround to have the 'Build plate adhesion' set to 'off', and nothing works.

Like the comment above, this is the workaround difference:

> Tab Anti Warping with skirt enabled

[image]

> Tab Anti Warping with skirt disabled

[image]

This also dramatically affects regular support stability with build plate adhesion set to 'off'. Here's an example of a support structure's first 3 layers. The blue layer shown here is the first layer. Subsequent support is printed mid-air.

[image]

hellodosi, 2023-06-16:
I have already gone crazy, because I have searched all settings to find the error. Then I hope that this is quickly fixed. until then I print just a skirt also. then it works.

discip, 2023-06-16:
@casperlamboo
Despite the fact that the brim was _improved_ this issue still remains!
Please fix that.

discip, 2023-06-16:
@casperlamboo
Despite the fact that the brim has been _improved_ in [5.4.0-beta1](https://github.com/Ultimaker/Cura/releases/tag/5.4.0-beta.1), this problem still persists!
Please fix this.

MariMakes, 2023-06-26:
Hey all,

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14983
We have a ticket on our backlog with the intent to improve this behavior. I'll raise it again with the team to see if we can increase the priority. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14983