Ultimaker/Cura #14811: strange artifacts in skin

discip, 2023-03-07:
### Application Version

5.3.0

### Platform

Manjaro

### Printer

unrelated

### Reproduction steps

Open the attached `3mf` and slice it with both **`5.2.2`** & **`5.3.0`**, compare the result by looking for **_tiny holes_** in the skin.

### Actual results

There are artifacts in the skin when sliced with the latest release.

### Expected results

No artifacts in the skin should occur, as in the previous release.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[project file](https://we.tl/t-42x3Y7UqdD)

# overview
## 5.2.2
[image]
## 5.3.0
[image]
<br>
<br>
# artifact 1
## 5.2.2
[image]
## 5.3.0
[image]
<br>
<br>
# artifact 2
## 5.2.2
[image]
## 5.3.0
[image]
<br>
<br>
# artifact 3
## 5.2.2
[image]
## 5.3.0
[image]

GregValiant, 2023-03-08:
Thanks for the report and the file.
Here is the model in MS 3D Builder.  You can see that it is reporting that there are errors in the model.
[image]

I used Cura 5.3beta and created a gcode from your project and read the file into AutoCad.  Here is that same area.
[image]
I measured the error and for both the larger marks it is 0.15mm.  They were the only glitches I see in the gcode.

Here is a view in Mesh Mixer.  You can see that the triangle density of the model varies widely from area to area.  This is not the smooth model that it appears to be at first glance.
[image]

In regards to this bug report:
The model has some definite issues and it's possible that 5.2.2 is simply more tolerant of model problems than 5.3 is. The slice I did with 5.2.2 does _not_ show those hiccups that are in the 5.3.1beta slice.

The Cura team will take a look

discip, 2023-03-09:
@GregValiant
Thank you for the investigation.
It may be that the object has issues and most likely this is the cause for the dimples, but as mentioned in the first post addendum, unfortunately that's not all!
## There are also these noticeable jagged paths on curved surfaces.

### The same strange jagged paths also appear on a smooth surface of a simple cylinder, while there is nothing of the sort in **`5.2.2`**.

Overall, the surface in **`5.2.2`** is much smoother compared to the same surface ​​in **`5.3.0`**.

# Please enlarge the images for a better view.

There is a very short extra path inserted between 2 consecutive segments of a curved path:

[image]

[image]

# You can see the difference particularly well on the left side.

[image]

[cylinder.zip](https://github.com/Ultimaker/Cura/files/10936528/cylinder.zip)

discip, 2023-03-14:
@MariMakes
Did anyone look into this yet?

Yhunoo, 2023-03-15:
I also saw these distinct jagged paths on the surface when slicing with version 5.3 ,

Yhunoo, 2023-03-15:
Does anyone solve this problem

aaron-crocker, 2023-03-18:
Im having the exact same issue with 5.3. Ive been racking my head for days on this one.  Smooth curved surfaces with dimples and what looks like extra seams forming vertically up the print.

discip, 2023-03-23:
@GregValiant
@MariMakes
@jellespijker

Is anyone on this already?

darrellenns, 2023-03-26:
I'm seeing this as well. I pasted my comments below from #14943, since this thread has more activity and it seems to be a duplicate issue.

I'm seeing the exact same thing. This is a regression in 5.3.0 - the problem does not appear in 5.2.2. It seems to happen when a wall goes from straight to large radius convex curve. There is a weird extra move in the gcode output. It's very hard to see in Cura's preview, but is quite clear if the gcode is viewed in the PrusaSlicer viewer.

Here is an example:
[image]

In Cura's preview, you can see a slight discontinuity:
[image]

The resulting print:
[image]

The same STL sliced with Cura 5.2.2 does not have these artifacts. Prusa Slicer also does not have this issue. Note that this is not the Z seam - that's in a different location. This artifact appears in 4 places on this model - every location where a straight wall connects to a large radius convex curve. It does not appear on the small radius curves (small fillets) or on the concave curves.

Here is the STL I used and resulting gcode:
[STL File](https://mega.nz/file/tc5VUZIS#LIH4um4DLUlyXwsaDhgTzVZCoBlZIm4K8e-69ub1Rik)
[5.3.0 gcode](https://mega.nz/file/tABAjArS#Ikspy-CrnoSEGB7TuZhF4yJPt4lbmFpPwtqq9lb88eY)
[5.2.2 gcode](https://mega.nz/file/gYhw0DIR#1FExquIXyAwgQOXUMzQsennbgpMRIM2JurKCpg6_Mg8)

Both gcode samples were sliced with the Cura default standard 0.2mm profile.

qwerty8224, 2023-03-28:
I reported this problem too, but I was sent to the other side )))) I found one way out is to repair objects to increase the maximum resolution. The problem is massive, but apparently they don’t want to listen to us, alas ((((

qwerty8224, 2023-03-28:
can anyone from the developers confirm the problem? or is that not a problem?

qwerty8224, 2023-03-28:
> @qwerty8224 Как заявили все комментаторы здесь, это определенно проблема!
> 
> Я думаю, что вы имели в виду, что вопрос об этой проблеме еще не решен.😊

they didn’t confirm the problem for me, they sent me to read YouTube and look for the cause in the printer 🤷

discip, 2023-03-28:
@qwerty8224
According to your description, including the solution: "Increase the resolution", your issue is not the same as this one.
Since applying your solution doesn't fix the problem I'm referring to here.

igiannakas, 2023-04-05:
Same issue here with curved surface models. Seems like the slicer is having a numerical precision issue when planning the path for curved surfaces.

darrellenns, 2023-04-06:
There does seem to be something profile-related going on here. I did the following as a test:

- Remove all Cura settings files
- Launch Cura 5.3.0
- In the first run dialogs, configure Cura for a Creality Ender 3 Pro
- Load same STL from my earlier post and rotate it 90 degrees for correct print orientation
- Select the Cura provided 0.2mm profile
- Slice the model and look in Cura's internal preview for the glitch

If I do those steps exactly, the glitch is there. As I mentioned before, spotting it in Cura's preview is a little tricky but it is visible. It's more obvious if you select the extruder preview control at the bottom and move one step at a time using the arrow keys on the keyboard.

I then tried the exact same steps, but using a different printer model in the first run dialog. I tried Prusa MK3, Ultimaker S5, and "Custom FFF". None of these exhibited the glitch. Additionally, I tried changing the machine/extruder settings for the Custom FFF to match the Ender 3 Pro settings. It still did not exhibit the glitch. I then tried a few different Creality printers, and all of them showed the glitch.

darrellenns, 2023-04-06:
After some further experimentation, I found the combination of settings that results in this glitch:

- Mesh Fixes -> Maximum Resolution: 0.25
- Walls -> Print Thin Walls: disabled

These are the default settings for Creality machines. In the "Custom FFF" machine, the defaults were resolution 0.5 and thin walls enabled.

So it seems there is a bug that is triggered by disabling "Print Thin Walls", but the glitch is filtered out if "Maximum Resolution" is set too high for the glitch to show up in the result.

darrellenns, 2023-04-06:
As a workaround, you can edit creality_base.def.json and remove the line with "fill_outline_gaps". This enables "Print Thin Walls" on all profiles for Creality machines, unless you have explicitly disabled it in your custom profile. Of course, creality_base.def.json will be replaced next time you upgrade Cura, but hopefully the bug will be fixed in the next release anyway.

igiannakas, 2023-04-06:
Not sure this works for me - I have a custom profile and I have thin walls enabled and the issue persists. It moves if I change the mesh resolution (I.e: the lines go to another place in the model) but doesn’t disappear.

MariMakes, 2023-04-06:
Hey All,

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10500

Thanks for the report! 👍

themuck, 2023-04-26:
> As a workaround, you can edit creality_base.def.json and remove the line with "fill_outline_gaps". This enables "Print Thin Walls" on all profiles for Creality machines, unless you have explicitly [...]

Thanks it works when i enable the "Print Thin Walls" function for my Cr-10, standard PETG and Cr-10 Printer Settings.

[image]

plainoldjim, 2023-04-28:
I've been having this issue. It seems to be related to infill somehow. I accidentally printed with 0% and the print came out perfectly. when I turned infill back on, the artifacts / warts came back.
[image]
[image]

alexdoni, 2023-04-30:
Same issue here with 5.3.1 (5.2.2 not a problem).
I did look into the gcode generated by 5.3.1 and it kinda duplicates some of the movements; X,Y have almost identical positions.
See some pictures bellow. 
[image]
[image]

discip, 2023-05-01:
@plainoldjim
> I've been having this issue. It seems to be related to infill somehow. I accidentally printed with 0% and the print came out perfectly. when I turned infill back on, the artifacts / warts came back.

Unfortunately I can't confirm that.
Tried what you suggested but the artifacts remained.

plainoldjim, 2023-05-01:
Very interesting. I will have to do more testing, but I've currently rolled
back to 5.2.2 to take care of a few jobs so it won't be tested on my
machine for a while.

discip, 2023-05-01:
> but I've currently rolled back to 5.2.2

That's exactly what I did! 😊
But exclusively for testing purposes I also left *`5.3.1`** installed!

discip, 2023-05-01:
> but I've currently rolled back to 5.2.2

That's exactly what I did! 😊

But exclusively for testing purposes I also left **`5.3.1`** installed!

PhilBaz, 2023-05-05:
Project file for stuttering zits on round surface....referenced above.
[5.3 stuttering zits.3mf.zip](https://github.com/Ultimaker/Cura/files/11409933/5.3.stuttering.zits.3mf.zip)

Wolfgang-create, 2023-05-18:
Same here, since Cura5.3.1 I got ugly vertical blobs on my outer wall as well.

DenTechs, 2023-05-19:
I also seem to be experiencing this issue, notice that when printing the toolhead would appear to stutter while going across a curve, but it clearly wasn't a hardware issue as it was retracting during those stutters as well. It ended up leaving artifacts in the surface from the drop in speed/flowrate
[image]

qwerty8224, 2023-05-19:
Yes, there is an obvious problem, I wrote about it from the release of version 5.3 already 5.3.1, but the problem was not solved (((can 5.3.2 be released? Well, it’s not possible to work in a broken version of the program (((

MariMakes, 2023-05-19:
Hey @qwerty8224,

We really want to fix this for the upcoming 5.4, A few of our developers have found the root cause, but the solution will require some time, so I can't make any promises. Apparently, micro-segments have been a problem for a while but with this release, they are aligning and it's become more obvious. 

Bugs that impact the print quality are taken seriously, so we hope to be able to update you soon.

Wolfgang-create, 2023-05-19:
Thank you very much for answereing , I really appretiate your hard work, as 
well as the good work of the developers.

I will stay tuned for upcoming releases.

Stay health, and a awesome weekend.

Best regards
Wolfgang

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<https://kommunikationsdienste.t-online.de/redirects/email_app_android_sendmail_footer>

--- Original-Nachricht ---
Von: Mariska
Betreff: Re: [Ultimaker/Cura] 5.3.0 strange artifacts in skin (jagged paths 
on curved surfaces) (Issue #14811)
Datum: 19. Mai 2023, 14:12
An: Ultimaker/Cura
Cc: Wolfgang-create, Comment

Hey @qwerty8224 <https://github.com/qwerty8224> ,

We really want to fix this for the upcoming 5.4, A few of our developers 
have found the root cause, but the solution will require some time, so I 
can't make any promises. Apparently, micro-segments have been a problem for 
a while but with this release, they are aligning and it's become more 
obvious.

Bugs that impact the print quality are taken seriously, so we hope to be 
able to update you soon.

—
<https://github.com/Ultimaker/Cura/issues/14811#issuecomment-1554481310> ,

aaron-crocker, 2023-05-20:
Thank you so much for all of your hard work and acknowledging the problem.
You guys are awesome, always!  Please keep the 3d printing community
posted.  Have an amazing weekend

Aaron Crocker-

PhilBaz, 2023-06-16:
Stuttering still present in 5.4 Beta
[Stuttering.zip](https://github.com/Ultimaker/Cura/files/11775442/Stuttering.zip)

PhilBaz, 2023-06-16:
@MariMakes Hi mari,
please see above

discip, 2023-06-16:
@casperlamboo
Unfortunately the claim of having solved this issue with the latest release is not true. 😞
The behavior is nearly unchanged

L0laapk3, 2023-06-18:
Is this something thats expected to be fixed in 5.4.0 still? Or has this been postphoned to the next minor or major?

jellespijker, 2023-06-19:
@discip, I'm sorry to hear that you still have this problem. I would like to ask you to share a bit more than your current statement: "nearly unchanged". We did work on this problem and we tried our best to limit the small path deviations that could (and can) still happen during slicing. Did you come to this conclusion with actual prints or by looking at the preview GCode?

The artifacts that occurred in 5.3 were caused by either multiple tiny paths placed after each other, which could cause a buffer underrun in some printers. Simply put, the printer couldn't handle a burst of GCode commands in a timely manner, or by tiny paths with a sudden deviation of the angle compared to an ideal fluid motion line.

During slicing, the generated polygons, and at a later point in time, the generated toolpaths are simplified. This will filter out tiny line segments as much as possible while still trying to hold true to the shape of the model and limiting the inadvertent generation of other print artifacts, such as banding. Banding is a slight shift of a longer path in position and orientation compared to the shape of the path in the previous or next layer.

The simplification looks at the following conditions:

If the deviation from a point is less than x amount from the distance perpendicular to the base of a line drawn from the previous point and the next point, then consider this point for removal. This is known as the maximum deviation in the front-end.
If the length of a line from the previous point to the current point is less than 5 microns, remove it always.
If the length of a line from the previous point to the current point is less than the maximum resolution, consider that point for removal.
If the point is to be removed, and it will affect the area of the triangle between the previous, current, and next point by deviating more than x amount, then don't remove the point. Since this will result in banding problems, which are even more visible than blobs and also impact the dimensional accuracy negatively.
The root cause of this problem lies in the fact that the simplification of our polygons and toolpaths is over-constrained because of the above-mentioned conditions. But we also know from experience that these conditions are needed. The simplification algorithm can therefore never find an optimal solution but will find a solution within an allowable range.

We worked on improving this problem in at least two sprints, with two developers, and we managed to improve it compared to what it was in 5.3. We see a drop in the violations of sequential GCodes compared to 5.3.1, as shown in the graph(s) below. Here we analyze the violations for the buffer underruns. Keep in mind that a single small line segment in itself might not be an actual problem, but a number of them might be.

[image]

This next graph shows that there are still a lot of violations that occur when looking purely at the theoretical data. This is mostly due to this particular model, which consists of a number of spikes where the diameter of each spike goes towards a diameter of 0.4 mm. Here you see that in order to hold true to the shape of the model, we simply can't create lines long enough such that we don't have any violations.

[image]
We keep on working on this problem, but unfortunately, the solution isn't as clear-cut as we want it to be and very complex in its nature. Mostly because simply analyzing the print in something like the gcodeanalyzer or in the preview view of Cura will only tell you that there is a small segment somewhere. It doesn't show if that small segment is actually a problem.
That will only be visible in actual prints, and for this, we could really use your help.
Because our UltiMaker printers don't show these defects.

If the problem still occurs, I would like to ask you guys to do some actual prints where you try out different resolution and deviation settings, to see if these settings are actually sensible for your printers.

And maybe help figure out which simplification conditions should weigh more in the simplification algorithm. Keep in mind that you can't have it all :sob:. Believe me when I say that I'm as frustrated by this as you guys.

Please share the project files and photos of the print that still show the problem so that we can use these for further analysis.

@Turbine1991 and @L0laapk3, we're trying to improve this even further in the beta period. But since it isn't a regression compared to 5.3, we will most likely continue with the stable release for 5.4 without further improvement. That doesn't mean that we will stop working on this problem, and if we manage to improve it a lot, then it will certainly warrant a patch release 5.4.1.

PhilBaz, 2023-06-19:
@jellespijker 

Please see my attached files above. 

Is the graph for infill the same as for outer and inner walls? I would think infill would be the most important part of the problem. We get zits on otter walls from the stuttering.  Maybe they are exactly the same result...I dont know.

The stutters manifest as zits so they are very easy to see through a simple print, I find the head of matterhackes Phil-A-ment exposes them perfectly.

I understand you have constraints, but Smartavionics master branch, Prusa, Orca etc. dont have this problem. So if this is the hill Cura dies on, then so bit it. It is a major make or break bug, along with a few others that have existed since 5.0 came out.

discip, 2023-06-19:
@jellespijker
### Thank you for the extensive explanation. 👍🏻
I do somewhat get what you explained.
However, this issue was previously not present (5.2.2).

1. You're right, I haven't printed with the latest beta yet.
But judging by the results visible in the slicer when comparing 5.2.2 and 5.4.0, there are big differences.

PhilBaz, 2023-06-19:
@jellespijker 

Yes, actual prints from that exact project file and Gcode. Also compared with Smartavionics branch, and of course the whole lot just went out to the trash. Ill print the beta again. 

You can see the resolution settings in the file. The same i use with SA branch. If you have a suggestion I can try it.

No worries about the graphs, was just wondering which one you were using for your findings as infill wouldn't be as useful as wall.

Also, I'm running all this code on a klipper machine with tons of processing headroom if that makes any difference. No 8bit buffer here.

PhilBaz, 2023-06-19:
@jellespijker Ps... you can also compare with a very similar file I attached in my may 5th comment above, if that's helpful.... its 5.3 with similar issue.

darrellenns, 2023-06-19:
@jellespijker

> The artifacts that occurred in 5.3 were caused by either multiple tiny paths placed after each other, which could cause a buffer underrun in some printers. Simply put, the printer couldn't handle a [...]

In the cases I observed and reported in this thread, it was definitely a "tiny paths with a sudden deviation of the angle" rather than a buffer underrun. It's a single line of gcode with a sudden angle change.

Do you have any idea why enabling "print thin walls" works around the problem? Does disabling print thin walls do something to do the path simplification algorithm that is introducing the problem?

harishrajan96, 2023-06-20:
I tried a benchy in 5.4.0-beta.1 it still causes the zits and blobs. Please check #15813 I have posted some pictures there.

jellespijker, 2023-06-20:
Thank you guys for your input, I will take it all into account when working on this. I might ping you guys on occasion during when I got it in a state that it needs some actual printing. Otherwise I'm working completely blind, since our printers Don't show this behaviour. I also can't make any promises that this will make it for the stable release, but if I finish after the stable I believe it will warrant a patch release.

Devs for internal reference see: CURA-10724

I'm also curious if you guys are running Klipper or Marlin.

L0laapk3, 2023-06-20:
> Thank you guys for your input, I will take it all into account when working on this. I might ping you guys on occasion during when I got it in a state that it needs some actual printing. Otherwise [...]

I was under the impression that this issue was mostly for klipper, or that may have been one of the issues closed in favor of this one :p

I can tell you that when I shared my print with zits in the klipper discord, someone immediately diagnosed it correctly as me having used cura 5.3, as far as I can tell it is the common consensus there that cura 5.3 generates unusable gcode for firmware that actually respects the generated paths to a high degree of accuracy (aka klipper)

harishrajan96, 2023-06-20:
> I'm also curious if you guys are running Klipper or Marlin.

I am running klipper, but I have tried this in marlin(+Cura 5.3.1), I see a proper Z seam in marlin that can be hidden

poulpor, 2023-06-20:
I use Klipper and Marlin on an Elegoo Neptune 3 Pro printer, and I'm happy to know it's a slicer bug and not a printer or filament problem, because I couldn't understand why my "curvy" prints were full of zits/blobs when using Klipper (no problem when I reinstall Marlin firmware with the same gcode print file). 
I started to suspect it was a slicer problem when I realized that the blobs/zits are always located at the same place when I printed the same 3D model at different speeds and settings for tests.
 
I currently use Ultimaker Cura 5.3.1, didn't try 5.2 yet.

[image]
[image]

cohaolain, 2023-06-21:
Just adding my two cents here: I'm also experiencing this with recent versions of Cura. The latest working version I've tried was 5.2.2, I first time I saw this issue was after updating to 5.3.1 .

I took some (slow-motion footage)](https://github.com/Ultimaker/Cura/assets/20997804/97680a66-65d2-41ea-bfca-13c3d1cff60e) of my printer in action, since I saw blobs/zits appearing relatively consistently along one side of print (_not_ the seam), to confirm what I thought I was seeing. I was able to confirm my printer was consistently slowing down significantly at a number of points along the curved surface.

Taking a look at the preview in Cura, I could not see anything peculiar at that point. However, when I saved the gcode and re-loaded it in Cura from there, the imperfections did appear in the preview, and they correspond to where I saw my printer slowing down [(video)](https://github.com/Ultimaker/Cura/assets/20997804/97680a66-65d2-41ea-bfca-13c3d1cff60e)
[image]

Finally, out of curiosity, I dug into the gcode to try to identify _why_ my printer was slowing down. There were no speed modifications or anything else that obvious. I wrote some quick python to graph the line movements around that area and show each movement's length and saw some very interesting outliers.

Graphing most of the main outer shell path here, and highlighting line segments <0.1mm:
[image]

We can easily spot the problematic sudden changes in angle which are forcing the printer to almost completely stop in its tracks for a brief moment, leading to the dreaded _blob_:
[image]
and
[image]

jellespijker, 2023-06-21:
thanks for all the good comments guys, they align with my findings as well.

I'm currently investigating the effects of adding a polygon/extrusion line smoothing operation after the simplification algorithm.

This should help smooth out sudden bigger changes in the paths. Which are more noticeable in smooth motion planners, such as Klipper.

This could result in us adding a smooth factor setting which will probably be printer (GCode dialect specific)

L0laapk3, 2023-06-23:
@jellespijker These particular sentences of your explenation have been haunting me for some time:

> The artifacts that occurred in 5.3 were caused by either multiple tiny paths placed after each other, **which could cause a buffer underrun in some printers. Simply put, the printer couldn't handle [...]

> We worked on improving this problem in at least two sprints, with two developers, and we managed to improve it compared to what it was in 5.3. We see a drop in the violations of sequential GCodes [...]

There appears to be some mixing up between two different phenomena, I think there is an important destinction to be made here:
1. Too many small gcode commands are sent, causing a buffer underrun and pausing the printer until new commands are received
2. A single tiny path with a sharp direction change forces the printer to slow down drastically to adhere to acceleration constraints

As far as I understand these are two totally different things. (obviously many of 2. may also cause 1.) By the sounds of it you guys have been mainly doing sprints hunting after 1. However I believe that 2. is actually what is causing problems here, and my understanding is that klipper is better at avoiding 1. while making more efforts to follow the faulty 2. paths as listed in the gcode.

I have been trying to definitively rule out buffer underruns on my klipper machine, but have been unsuccesful in figuring out how to do so. But as far as I can tell, even a single faulty (= sharp angle change) line segment can often cause zits which strongly hints at 2 in my eyes.

jellespijker, 2023-06-23:
that is why I'm currently working on the smoothing part atm

jellespijker, 2023-07-01:
Hi All,

> TLDR; I believe I have managed to fix the issue (or at the very least, improve the behaviour). I need you guys to test it on actual printers since I cannot validate the fix (or even the bug) on our printers. Please do this at  your earliest convenience, because we plan to release the stable version at the beginning of next week and it would be a shame if this fix is not included.

# Problem

The simplification of polygons sometimes leaves small segments with large differences in angles compared to the segments before and after them. This occurs because the simplification algorithm is over-constrained, and the solution space doesn't diverge to an optimal point, but to a plateau. This means that smaller segments won't be filtered out since doing so would violate other criteria, such as removing too much area, etc.

Printers with Marlin 1.x seem to handle the resulting toolpaths better than smooth motion planners such as Klipper. These printers see a sudden change in direction and will try to adhere to the requested toolpath as much as possible. Because that toolpath is now suddenly requesting a change in direction (for a very short length), the head will slow down to make the corner, but the extrusion process can't adjust at the same speed, resulting in over-extrusion at that spot.

# Fix
Since we can reasonably assume that such a sudden change in toolpath direction over a short distance, compared to the direction of the toolpath before and after, does not belong to the actual intended shape of the sliced model, we can adjust the polygons in favour of the intent and smooth motion printers.

We loop over each closed polygon, which is used to generate the Aranche toolpaths, and examine 3 segments (4 points) each time: `AB` -> `BC` -> `CD`. Sudden small segments `BC` (< maximum resolution) that deviate more than a certain angle compared to a fluid motion angle (wall transition angle) between the preceding segment `AB` or subsequent `CD` are smoothed out. This is done by either shifting point `B` or `C` towards `A` or `D` by half the maximum resolution's distance or removing them altogether if there is no room to shift them.

[image]

# Process
We plan on releasing the stable version at the beginning of next week, regardless of whether this issue is actually fixed. Do not worry, work will continue on this issue and if we don't manage to include it in the stable version, we'll release a patch version. We decided on this approach to ensure that this ticket isn't rushed and our quality standard is maintained. We also need to release due to UltiMaker's new PET-CF materials being available (it would be odd if people can buy it in the shop but not slice it with Cura).

The Pull Request https://github.com/Ultimaker/CuraEngine/pull/1891 still needs to be reviewed and component tested by one of my colleagues (probably @casperlamboo) and QA tested by our QA engineers @HellAholic and/or @Vandresc. We need your help (@discip, @cohaolain, @poulpor, @aaron-crocker, @harishrajan96, @L0laapk3, @darrellenns, @PhilBaz, @Turbine1991, @Wolfgang-create, @qwerty8224, @DenTechs, @plainoldjim, @alexdoni, @themuck, @Yhunoo). We can't reproduce this issue on our own printers (otherwise, QA would have never let this pass to a release), and we don't own third-party printers (since Cura supports over 500 different printers from competitors). **We need you guys to test this proposed fix**, not just slice and preview the G-code, but actually print with this nightly and compare it with the 5.4.0-beta.1 or 5.3.x.

I have created a nightly: `5.4.0-beta.2` They can be downloaded here:

- [UltiMaker-Cura-5.4.0-beta.2+305e00-linux-AppImage](https://github.com/Ultimaker/Cura/suites/13996973703/artifacts/780981203)
- [UltiMaker-Cura-5.4.0-beta.2+305e00-linux-modern-AppImage](https://github.com/Ultimaker/Cura/suites/13996973703/artifacts/780981204)
- [UltiMaker-Cura-5.4.0-beta.2+305e00-win64-exe](https://github.com/Ultimaker/Cura/suites/13996973703/artifacts/780981205)
- [UltiMaker-Cura-5.4.0-beta.2+305e00-win64-msi](https://github.com/Ultimaker/Cura/suites/13996973703/artifacts/780981206)
- Mac ( Coming soon, the compiler Apple-Clang v13 is missing some functionality)

_I noticed that in atleast one test project file of mine a hard crash on the engine occured, not directly related to my fix and in a location where it previously did not crash. I will investigate further on Monday why that is. But the nightlies as they are now should at least validate the fix._

cohaolain, 2023-07-01:
Sounds promising, I'll take a look at this. I may not be able to print something soon, but I'll definitely be able to compare Cura's output with my python script to see if the short line segments have been eradicated. Essentially, this will check all line movements generated to see if any are shorter than 0.05mm. 

They are quite readily visible, occuring in groups, visible particularly when graphing the issue with many layers on top of one another - or when printing, and seeing them grouped together!
[STL zip file of affected file from Cura 5.3.1](https://github.com/Ultimaker/Cura/files/11926370/20230529_Rubber_Duck_02mmlh%40150mmps.flow%40100.gcode.zip)
[image]

This is the script I'm currently using. You can call it like this:
```bash
python test_gcode.py 20230529_Rubber_Duck_02mmlh@150mmps.flow@100.gcode 10 15
```
[test_gcode_script.zip](https://github.com/Ultimaker/Cura/files/11926418/test_gcode_script.zip)
Which shows all outer wall extrusion lines, highlighting short line segments (under 0.05mm), at layer heights 10-15mm.
With the affected gcode I've attached, this yields the following:
[image]
[image]

I'll take a look at this and compare with the new build soon :) 

Smoothing the output sounds like a good idea in theory, though I am still curious if this is just a temporary workaround rather than a permanent solution (better in the short term than not addressing, of course)? In particular, do we have any idea what change actually introduced this issue to 5.3.0? 

Thanks all for your time!

cohaolain, 2023-07-01:
A quick test still shows some of these bad segments appearing in gcode, but much fewer - 19 vs 102 at layer heights between 10-15mm.

Still seeing e.g.
[image]
[image]
[image]

STL, gcode: [duck_gcode_and_stl.zip](https://github.com/Ultimaker/Cura/files/11926432/duck_gcode_and_stl.zip)

cohaolain, 2023-07-01:
To summarize, it looks like the symptoms have been partially relieved, but the underlying issue is still apparent to some degree. Although I haven't tried printing this output myself, it seems apparent this would still have some marked impact on the final print quality.

jellespijker, 2023-07-01:
Can you share the 3mf instead of the stl since the settings and printer definitions determine a lot in the smoothing algorithm. 

b.t.w good approach using Python for this. We recently added some extensive logging to engine with our Visual Debugger and in my test projects I don't have any segments smaller then 50 micron between longer segments. Mind you they can still occur in sequence for very small radii. But then it is expected and acceptable since it is the intent.

[image]

jellespijker, 2023-07-01:
That being said, I do see a lot more violations when I slice that rubber ducky stl on a Creality printer. Which is a bit of a bummer, I will investigate further on Monday. Since I didn't intend to work this long on this ticket on a Saturday.

[image]

cohaolain, 2023-07-01:
Haven't seen ParaView before, looks cool. I'll share the .3mf when I'm home, enjoy your Saturday :)

cohaolain, 2023-07-01:
3mf as promised, the gcode from this on the new build yields 19 erroneously short line segments between layer heights 10-15mm, as mentioned above. The model is scaled 150%.
[test2_20230702_Rubber_Duck_02mmlh@150mmps.flow@95.3mf.zip](https://github.com/Ultimaker/Cura/files/11927707/test2_20230702_Rubber_Duck_02mmlh%40150mmps.flow%4095.3mf.zip)

jellespijker, 2023-07-02:
I did some quick fixes, eliminating a lot of those pesky segments. Two of the biggest culprits were an optimization made to check the deviation between angles (fix here: https://github.com/Ultimaker/CuraEngine/pull/1891/commits/f9a35382b0e4ec9dbef6e6fcd81fc58e268cbb62 ) and checking the first segment in the closed loop (fix here: https://github.com/Ultimaker/CuraEngine/pull/1891/commits/f694ca4bf4e7df509ea4ed63941d219c382de8bd )

I build a couple of new installer:
- [UltiMaker-Cura-5.4.0-beta.2+305e00-linux-AppImage](https://github.com/Ultimaker/Cura/suites/14007575333/artifacts/781763961)
- [UltiMaker-Cura-5.4.0-beta.2+305e00-linux-modern-AppImage](https://github.com/Ultimaker/Cura/suites/14007575333/artifacts/781763962)
- [UltiMaker-Cura-5.4.0-beta.2+305e00-win64-exe](https://github.com/Ultimaker/Cura/suites/14007575333/artifacts/781763963)
- [UltiMaker-Cura-5.4.0-beta.2+305e00-win64-msi](https://github.com/Ultimaker/Cura/suites/14007575333/artifacts/781763964)

[39 more replies]

closed 2023-07-05 by jellespijker