Ultimaker/Cura #14812: Small Layer Printing Temperature

hrdcpy, 2023-03-07:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Creality Ender - 3 / Ender - 3 V@

### Reproduction steps

1. Imported a 3mf file made in an older version of Cura 5.X.X that worked fine
2. Deleted model 
3. Placed new model STL
4. Hit "play" to slice the model

### Actual results

**Error Message**: ! Unable to slice
Unable to slice with the current settings. The following settings have errors: "Small Layer Printing Temperature"

_The text field is highlighted in RED_
_I've never modified this setting before_
[image]

### Expected results

The file should have been sliced and previewed as normal

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The 3mf file
[CE3E3V2_Headset_Hanger_v1.zip](https://github.com/Ultimaker/Cura/files/10914636/CE3E3V2_Headset_Hanger_v1.zip)

GregValiant, 2023-03-08:
Thanks for the report.
I wasn't aware this setting existed.  Here are a couple of lines from the fdmprinter file.
     "value": "material_print_temperature",
     "enabled": "cool_min_layer_time > 0",
    "minimum_value": "max(material_final_print_temperature, material_initial_print_temperature)",
    "maximum_value": "material_print_temperature",
With my print temperature at 210, Initial Printing Temperature 205, and Final Printing Temperature at 200 I can set the Small Layer Print Temperature between 205 (the larger of the other two settings) and 210 (the Print Temperature).  When I sliced a model with those settings, near the end of the gcode Cura inserted M104 lines and the temperature stepped down incrementally from 210 to 205 over the course of the last few layers.

In regards to this bug report - It appears that if the Initial Print Temperature, Print Temperature, and Final Print Temperature aren't configured correctly (the boxes turn yellow) that it causes Small Layer Print Temperature to be in error and Cura won't slice.  I don't know why it's set up that way but it makes it appear that what you have is a well hidden "expected behavior".
The Tooltip text doesn't mention the connection with Initial or Final print temperature and it appears that it should.
I think the Cura team is busy getting ready to release 5.3.2 but maybe @MariMakes or someone else from the 
team can chime in on this.

hrdcpy, 2023-03-08:
The original 3mf file was probably saved in Cura 5.3.0-beta.2 and that file
sliced as expected and the gcode printed successfully.

MariMakes, 2023-03-08:
Hey @hrdcpy,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍 We've been seeing similar reports.

This, Small Layer Printing Temperature, is a newly introduced setting in Cura 5.3.
It was assumed that nobody would raise their initial printing temperature above the final printing temperature. 
But we see more reports of people doing this. 
[image]

Instead of blocking people from doing this, it's better to change it into an error like we do with similar settings. 
I'm discussing with the team if we can do it with everyone, but this is what you can do:

Easy option 1: lower your initial printing temperature. 
Option 2: Change your max/min into a warning. 
- Go to the following folder C:\Program Files\UltiMaker Cura 5.3.0\share\cura\resources\definitions
- To find the following file fdmprinter.def.json
- Find the Small Layer Printing Temperature definition
- Change  "minimum_value" to "minimum_value_warning" and "maximum_value" to "maximum_value_warning" 
- Save the file and restart Cura to slice

Then you'll see this. 
[image]

Thanks for the report! It really helps us to understand how Cura is used.

kandrejevs2, 2023-03-08:
> It was assumed that nobody would raise their initial printing temperature above the final printing temperature.

that's actually super common to get better bed adhesion to lay initial layer hotter.

Asterchades, 2023-03-09:
That's what the `Printing Temperature Initial Layer` setting is for. The `Initial Printing Temperature` sets the M109 (heat while waiting) value added, before an M104 (heat but keep going) is added to the `Printing Temperature Initial Layer` so it can continue heating while still processing the start GCode.

Though having two similarly-named settings perform such different purposes has undoubtedly been problematic for some users - especially with the interaction to other settings not being readily exposed in the UI.

MariMakes, 2023-03-13:
This should be resolved for everyone in an upcoming patch release Cura 5.3.1
The change was made here:  https://github.com/Ultimaker/Cura/commit/04c6f6bbcf31df0cd70c424c08f76b918cf1d4eb

Thanks again for reporting! 👍

closed 2023-03-13 by MariMakes

hrdcpy, 2023-03-13:
Thanks to you all. I'm going to practice better file hygiene going forward.