Ultimaker/Cura #14813: Auto drop down always on

the-minist3r, 2023-03-08:
### Application Version

5.3.0

### Platform

windows 11

### Printer

ender 3 pro

### Reproduction steps

Adding models automatically drops them down to the build plate when slicing even with auto drop down turned off and the box on the left unchecked. Tried turning on then back off with no change.

### Actual results

slices normally but drops the model down as if automatic drop down is turned on.

### Expected results

model should stay suspended and support generated underneath

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Egg.zip](https://github.com/Ultimaker/Cura/files/10915339/Egg.zip)

MariMakes, 2023-03-08:
Hey @the-minist3r,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

We have a checkbox in the preferences to define your preference.
You can find it under Preferences | Configure Cura | Automatically Drop Models to Buildplate:
[image]

To me, it doesn't seem like a bug in Ultimaker Cura since there is a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

the-minist3r, 2023-03-08:
I have that turned off as well as the check box on the left side.  I can
record the issue later and add that to the bug ticket if necessary.

the-minist3r, 2023-03-09:
Here is a short video showing the relevant settings and the issue I'm
describing.  I want the eggs to stay suspended so that the support prints
under them instead of the bottom of the egg being on the build plate.  The
finish is much better on round models with this method of printing and is
necessary in instances like this.

the-minist3r, 2023-03-09:
[auto drop down.zip](https://github.com/Ultimaker/Cura/files/10926584/auto.drop.down.zip)

Here is a short video showing the relevant settings and the issue I'm describing.  I want the eggs to stay suspended so that the support prints under them instead of the bottom of the egg being on the build plate.  The finish is much better on round models with this method of printing and is necessary in instances like this.

maruhe, 2023-03-10:
i have same behavor since 5.3 release - lucky you if you didn't crash your bed
[#14828](https://github.com/Ultimaker/Cura/issues/14828)
I'd please you to enable adaptive layers, export the gcode and check if it starts at Z below 0
seems like you have support-roof active already? if not pls give it a try - would be nice if someone else also could reproduce GCode starting at negative Z 

oh, and a workaround if you like to print your eggs now - just place a litte object anywhere on Z=0 and all works fine

the-minist3r, 2023-03-10:
> Hey @the-minist3r,
> 
> Thanks so much for you video.
> 
> I was able to follow your exact steps but my result is different 😮 [image]
> 
> Could it perhaps be related to a plug-in that you have loaded? Like the Auto Orient plug-in?

Uninstalled the AutoOrient plugin and I'm still having the same issue.

the-minist3r, 2023-03-10:
> i have same behavor since 5.3 release - lucky you if you didn't crash your bed [#14828](https://github.com/Ultimaker/Cura/issues/14828) I'd please you to enable adaptive layers, export the gcode and [...]

Adding another object at Z0 did the trick to force support under the other models.  I haven't had any issues with the Z being below the build plate though even with adaptive layers.

fieldOfView, 2023-03-10:
This has nothing to do with object dropping in the front-end, but  with the "Remove Empty First Layers" setting which acts on the sliced result. You may try turning that off (it is on by default to save material).

maruhe, 2023-03-10:
nice input @fieldOfView,
"Remove Empty First Layers" gehts hidden when support is active - seems like it gets disabled also in older Versions (until Beta2) but is hidden and active in 5.3 release

the-minist3r, 2023-03-10:
> nice input @fieldOfView, "Remove Empty First Layers" gehts hidden when support is active - seems like it gets disabled also in older Versions (until Beta2) but is hidden and active in 5.3 release

I had to turn off supports, turn off remove empty first layers, save the profile, then turn supports back on to get it to act right.  Default behavior should be to turn it off when "Automatically drop models to the build plate" is turned off, regardless of support settings.

the-minist3r, 2023-03-10:
FFS. Now support is causing new issues where it doesn't start on the first layer. 5.3.0 is turning out to be a hot mess.
[support issue.zip](https://github.com/Ultimaker/Cura/files/10946608/support.issue.zip)

Support floor doesn't change if it starts on layer 1 or 2 either but I may need to submit a new report for this one.

the-minist3r, 2023-03-10:
[image]
[image]
FFS. Now support is causing new issues where it doesn't start on the first layer. 5.3.0 is turning out to be a hot mess.

Support floor doesn't change if it starts on layer 1 or 2 either but I may need to submit a new report for this one.

maruhe, 2023-03-10:
you need to have general skirt or brim active, it's a known bug

fieldOfView, 2023-04-07:
> either the "Remove Empty First Layers" shouldn't be active if it's not visible

I could not disagree less with that option. Setting visibility does not affect setting behavior for any other setting, so singling one out a specific setting to do that is a very bad idea.

slabua, 2023-04-07:
I can't find the Remove Empty First Layers setting anymore, although the search result shows that it is indeed supposed to be under the Mesh Fixes section, despite I have either skirt or brim active.
Which other setting controls the visibility of Remove Empty First Layers?

fieldOfView, 2023-04-07:
Currently the "Support" setting needs to be disabled to be able to change the "Remove Empty First Layers" setting. This may be considered a bug.

fieldOfView, 2023-04-07:
Sorry, I meant "Generate Support"

slabua, 2023-04-07:
Ah thanks :s

MariMakes, 2023-05-16:
Hey! 

Quick update on our side. 😊 
During clean-up, we found that this report seems to be a duplicate https://github.com/Ultimaker/Cura/issues/15179

And... Good news, it's resolved with the next release with this fix: https://github.com/Ultimaker/CuraEngine/commit/3b1f2e612534f0ed05e022159a5308a6d2c30479

🎉  I'll be closing this issue since there is a solution in place.
Thanks again for the report!

closed 2023-05-16 by MariMakes