Ultimaker/Cura #14815: Skirt/Brim outside of build plate boundaries

ThatOneKidLiam, 2023-03-08:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ender 3 S1 Pro

### Reproduction steps

1. Skirt or brim enabled
2. Object is moved to where skirt or brim is now outside of build plate boundaries 

### Actual results

Skirt or brim is sliced correctly (along with object), though the part of the brim or skirt that is outside of build plate boundaries is now cut off

### Expected results

Cura should give a warning saying that it will not slice correctly

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm not sure if this is an intended feature or not, I just wanted to report this just in case its a bug that needs to be fixed. I have not tested to see if this will damage the printer or cause issues (have not saved gcode and printed file).

I'm new to github and am not sure how this platform works, so I'm hoping this works.

GregValiant, 2023-03-08:
Thanks for the report.
I can duplicate this bug.
This is in 5.2.2.  The model is too close to the edge and the "reserved for brim" gray area is present.
[image]

This is in 5.3.s-beta and you can see that the brim reserved area is ignored and the model was allowed to slice.
[image]

I'll mark this as a Slicing Error (it should not have sliced) and @MariMakes of the Cura team will take a look.

MariMakes, 2023-03-08:
Hey @ThatOneKidLiam,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This seems to be one of the new features introduced in Cura 5.3
You can read more about it here: https://github.com/Ultimaker/CuraEngine/pull/1613
It's done on purpose to allow more print jobs on your buildplate/

If you print it, does your printjob fail? And if it does, can you show a picture?
Then we can take another look at the implementation.

But to me, it doesn't seem like a bug in Ultimaker Cura. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-04-21 by github-actions[bot]