Ultimaker/Cura #14818: 5.3 Linux at each start the welcome screen (and procedure) appears.

Rudi56, 2023-03-08:
### Application Version

5.3

### Platform

Linux (Opensuse)

### Printer

Elegoo Neptune 3 pro

### Reproduction steps

I start the "UltiMaker-Cura-5.3.0-linux.AppImage" first time.
All Settings of my 5.2.2 version is taking over.
After a exit  "UltiMaker-Cura-5.3.0-linux.AppImage" and 
restart "UltiMaker-Cura-5.3.0-linux.AppImage"
the Welcome screen appears and i must add a printer.

### Actual results

the Welcome screen appears and i must add a printer.

Every start the same procedure.
After adding a new printer, cura will display all old printers an I can select my printer an can work.
Next start the same procedure.

### Expected results

Normal start without add a printer every start.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No more additionals

ckvsoft, 2023-03-08:
Try the modern image. This works for me.
At least on Ubuntu

Rudi56, 2023-03-08:
I found the error.
An old version of a printer I used prior, had an error. 
(but cura doesn't tell me about them. :( )
I removed the old configuration files from that printer, so cura start with the "What news" screen now.
I think, The issue can be closed.

MariMakes, 2023-03-17:
Hey @Rudi56,

Sorry, it took us a while to get back to you 😞 
Thanks for the update. I'm Glad to hear you rissue has been resolved. 

I'll close the issue 👍

closed 2023-03-17 by MariMakes