Ultimaker/Cura #14819: Slicing Failure

RodislavR, 2023-03-08:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Loaded multiple objects on the built plate with my previous configurations and pressed Slice.
2. Repeated a few times. Still the same result.
3. Changed placement of the objects. Still same result.
4. Deleted objects. Still same result.
5. Loaded 3 small objects. Still Failure.

### Actual results

1. Slicing failed multiple times.
2. Changed to Cura 5.2.1 and Sliced my models. Worked out fine.

### Expected results

Cura should generate a G-code file. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

MariMakes, 2023-03-08:
Hey @RodislavR,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

 I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

RodislavR, 2023-03-08:
[Slicing Failed.zip](https://github.com/Ultimaker/Cura/files/10922727/Slicing.Failed.zip)

Thank you for the fast reaction! 

I have attached the zipped Project file. 

Kind regards,
Rodislav

MariMakes, 2023-03-09:
Hey @RodislavR,

Thanks for the file! I figured out what is going wrong. 👍 
Your Minimum Feature Size is too small. 
If you increase it to 0.1 your models slices again.
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore, since it's related to your settings. 
Would you still like to keep this issue open or do you agree that we can close it?

MariMakes, 2023-03-09:
Awesome! Good luck printing these cutters 🎉

closed 2023-03-09 by MariMakes