Ultimaker/Cura #14821: Way too many supports being generated

derekslenk, 2023-03-08:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Creality Ender 3 

### Reproduction steps

Uploaded a model to slice: https://www.thingiverse.com/thing:2683032
Turn on supports and slice.

### Actual results

Supports are generated for the threads only, lik eit did in 5.3.0-beta

### Expected results

Way tooo many support

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[CE3E3V2_DnD_Dice_Box_Base.zip](https://github.com/Ultimaker/Cura/files/10921510/CE3E3V2_DnD_Dice_Box_Base.zip)

MariMakes, 2023-03-08:
Hey @derekslenk,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This support looks correct to me 😅 
If you check under your model, you can see that there are a lot of red pats that need support. 
[image]

If you don't want those parts to be supported, and only the lid. 
You can change settings like Support Overhang Angle to reduce the support that is generated. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Does this solve your problem and can we close the bug?

derekslenk, 2023-03-08:
Yeah, that will work, I just was kind of shocked by default that the beta and new version generated such different support structures

closed 2023-03-08 by MariMakes