Ultimaker/Cura #14823: Please make select a face to align to build plate actually functional!

KevFinalRev, 2023-03-08:
### Is your feature request related to a problem?

I cannot get the align face to buildplate function to work at all. I click on a perfectly flat face and it aligns to some random edge.  I feel like this feature used to work in older versions of Cura from several years ago but has not worked in this version. 

### Describe the solution you'd like

Please make this feature work like it does in PrusaSlicer. When you select that feature, it creates a plane on possible face options to align to the buildplate. This gives you something to click on and it works every time.

### Describe alternatives you've considered

manually rotating the part has worked so far but some parts are complex with faces that I need to be aligned that are not at known angles from the initial position.

### Affected users and/or printers

everyone with any printer will benefit from getting this feature functional again.

KevFinalRev, 2023-03-08:
For reference, I am using Cura 5.2.2

5axes, 2023-03-08:
you should share an example because for me it works fine so could be an issue with your part or your harware.

KevFinalRev, 2023-03-08:
Well, I tested it with the part I was having specific problems with yesterday and it did what it was supposed to do. I think what I would like to see the addition of extra planes defining possible faces like they have in PrusaSlicer. I think that makes it much clearer. I am curious as to why I was having such trouble with this yesterday. I also had Cura crash on me multiple times yesterday and I believe it was due to a graphics overload. I wonder if those graphics issues were causing the problem by not properly resolving the faces and causing them to not be "flat". Oh well, you can close this request if you like. Sounds like it was a computer problem and not a Cura problem. Sorry about that.

Thank you!
-KevFinalRev

This e-mail message, including any attachments, is intended only for the use of the individual or entity to which it is addressed and may contain information that is proprietary, and/or confidential, and may be subject to export controls and restrictions including, but not limited to, the International Traffic in Arms Regulations (ITAR), the Export Administration Regulations (EAR) or the sanctions administered by the Office of Foreign Assets Control (OFAC). If you are not the intended recipient, please contact the sender and destroy all copies of the original message. Disclosure to foreign persons without prior U.S. Government approval is prohibited. Any unauthorized review, use, disclosure, or distribution is prohibited and may be subject to criminal prosecution.

ASLE1111, 2023-03-25:
i allso have this problem, with multiple different parts

logiclabs, 2023-04-05:
Same issue here with all my models. Click on a face and it aligns to a random edge. Using Intel UHD 770 Graphics.

KevFinalRev, 2023-04-05:
I now have Cura 5.3.0 and it still has this same issue. Sometimes it works, most times it does exactly what these other two people have mentioned. I click on a face, and it aligns to a random edge. usually not even a face. It picks an edge and rotates to align to that. including a feature that either highlighted the face our mouse hovers over or shows an arrow or creates planes to click on would at least allow us to know for sure that the software confirms our particular selection instead of us clicking on a face and not knowing why it aligns to an edge. If we can eliminate the bug altogether, that is also fine, but making it look similar to Prusaslicer would be nice.

massis87, 2023-04-21:
I had the same issue on Cura 5.2.2 and now on 5.3.1. Clicking a face to align on seems to randomly align to an edge, multiple attempts at aligning to the same face result in random orientations that differ every time.

massis87, 2023-06-19:
Upgraded to the new Cura 5.4.1 beta, the issue is still present unfortunately. And since I have a couple of models that were designed right way up but should be printed on their side (which is not at 90° to the default orientation), getting them laid out correctly in the slicer is incredibly annoying.