Ultimaker/Cura #14826: 5.3.0 closes small holes (5.2.2 did not)

mennucc, 2023-03-09:
### Application Version

5.3.0

### Platform

Ubuntu

### Printer

Creality Ender 3 Pro

### Reproduction steps

I have a model called [negative test](https://www.thingiverse.com/thing:4806918) with small slits, that can be used to calibrate the flow. 

With Cura up to 5.2.2 , it is sliced correctly, as you see in this image
[image]

With Cura 5.3.0, the smaller slits are filled in, as you see here
[image]

I have set 
- material: generic PLA
- printer : Ender 3 Pro
- settings : standard setting , with no changes

### Actual results

The slits are filled in, so the model is printed incorrectly.

### Expected results

The slits should not be filled in.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

For your convenience, here is the STL of the model: [negative_test.stl.zip](https://github.com/Ultimaker/Cura/files/10928725/negative_test.stl.zip) .

And the project file
[cura_5.3.0_.3mf.zip](https://github.com/Ultimaker/Cura/files/10928838/cura_5.3.0_.3mf.zip)

GregValiant, 2023-03-09:
Thanks for the report and the files.
Turn on "Print Thin Walls" and set the Minimum Feature Size to 0.05.
[image]

This looks like expected behavior but I'll leave the bug label on it  In this case it would appear that "Print Thin Walls" should be "Print Thin Gaps".  This is a well hidden phenomenon.

As an aside...
To calibrate "Flow":
Calibrate the E-steps and stop right there.  Use a magnifying glass to inspect a top skin and tweak the Cura Flow settings as appropriate per what your eyeballs actually see.  Nothing else is required or needed.

mennucc, 2023-03-09:
I can confirm that those two settings enable a correct slicing.

As for the "flow" calibration. I could not print some _print-in-place_ objects , such as [ kaleidocycle](https://www.thingiverse.com/thing:4761150) where there are very tiny gaps that must be empty ; my negative test was useful to set correct parameters.

GregValiant, 2023-03-09:
I'm going to re-open this.  I think it's something that isn't clear in Cura.  Maybe adjusting a Tooltip or something similar would help as "Minimum Feature Size" and "Minimum Thin Wall Line Width" are hidden unless "Print Thin Walls" is enabled.

UBaer21, 2024-02-24:
I just wondered about this ( Cura 5.6.0) as without "Print Thin Walls" enabled it seems the "Minimum Odd Wall Width" also determines the minimal gap/slit size which doesn't make sense.  When printing with big nozzles (like .8) you may not want to print a .4 wall but you sure can print a .5 gap.