Ultimaker/Cura #14827: 5.3.0 No profiles matching configuration of this extruder

kallisti5, 2023-03-09:
### Application Version

5.3.0

### Platform

ArchLinux

### Printer

Ultimaker 2 with Olsson Block

### Reproduction steps

1. Upgraded to Cura 5.3.0 from 5.2.2
2. Begin seeing "No supported profiles"
3. Unable to Slice

### Actual results

1. Unable to slice models in Cura 5.3.0.  Downgraded to Cura 5.2.2

### Expected results

It to work

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm not even sure why Cura cares about materials... my Ultimaker 2 has material selection built-in

kallisti5, 2023-03-09:
It looks like a hidden setting "Maximum Fan Speed" is defaulting to 15.0% (lower than Regular Fan Speed).

I erased the "Ultimaker 2 with Olsson Block" printer and recrated it.  While the profiles still show "Not supported", it at least slices again.

MariMakes, 2023-03-13:
Hey @kallisti5,

Thanks for your report 👍

I need a little bit more information to help you. 😅 
When I add an Ultimaker 2 with Olsen Block I don't have the option to change materials, just to change the printcores. 
Ultimaker 2s handle materials differently, so it could very well be that there are no materials profiles available. 

[image]

Can you share how you've added your printer to Cura?

closed 2023-04-21 by github-actions[bot]