Ultimaker/Cura #14828: Bed Crash - Cura 5.3 is starting at Z-1.19 when Model starts on support

maruhe, 2023-03-09:
### Application Version

5.3 release

### Platform

Windows 10

### Printer

Voron 2.4

### Reproduction steps

I placed a Model on Z=2 so the full bottom starts on support
tree-supprt is used and support-roof is enabled

### Actual results

already in slicing-preview I can see the the model get's dropped to the bed and no support is visible below but in gcode there is support below the full model but the support strarts at Z-1.19
[image]

### Expected results

first layer should strart above the print-bed
if I place a little object on the build-plate next to my model all is fine again

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

project-file is to big to upload it here, but I already posted in cura forum: https://community.ultimaker.com/applications/core/interface/file/attachment.php?id=97556&key=285cbeb3908738fbf7cc8ab04df7d3f2

gcode starts like that:
;Generated with Cura_SteamEngine 5.3.0
M82 ;absolute extrusion mode
;Nozzle diameter = 0.4
;Filament type = ABS
;Filament name = Extrudr ABS
;Filament weight = [16.105784921875003]

;M190 S100 ; Set bed temperature and wait
;M109 S260 T0 ; Set nozzle temp and wait

PRINT_START BED=100 EXTRUDER=260 MATERIAL=ABS MATERIAL_BRAND=Extrudr
G92 E0
G92 E0
G1 F1800 E-0.5
;LAYER_COUNT:396
_layer_change layer=0
;LAYER:0
M107
G0 F6000 X206.319 Y233.911 **Z-1.19**
;TYPE:SUPPORT
G1 F720 E0
G1 F600 X207.372 Y233.145 E0.06395

maruhe, 2023-03-09:
meanwhile I was able to reproduce with voron 2.4 standard profile
it always happened that the model get's dropped to the bed (or the full gcode gets "moved down") but only after I enabled adaptive layer height Z starts negative
[image]
project and gcode:
[V350_MGN9_Assembly_Tool.zip](https://github.com/Ultimaker/Cura/files/10936082/V350_MGN9_Assembly_Tool.zip)

maruhe, 2023-03-10:
"Remove Empty First Layers" gehts hidden when support is active - seems like it gets disabled also in older Versions (until Beta2) but is hidden and active in 5.3 release

when I disable support, than disable "Remove Empty First Layers" and then activate support again the GCode looks as espected and starts above the bed

maruhe, 2023-03-11:
got it done the other way around: the only model touching the build-plate was changed to mesh-type = support and gcode started first layer at Z0.71 (while first layer height is 0.2)
disabling "Remove Empty First Layers" fixed it

MariMakes, 2023-03-31:
Hey @maruhe,

Sorry, it took us a while to get back to you 😞 
To me, it doesn't seem like a bug in Ultimaker Cura anymore, since there is a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?