Ultimaker/Cura #14829: 5.3 Excessive support

Barry4ad, 2023-03-09:
### Application Version

5.3

### Platform

Mac OS 13.2.1 Mac Mini M1

### Printer

Creality Ender 5 Pro

### Reproduction steps

1. Loaded stl file (Blender 3.5)
2. Chose generate support - everywhere/normal
3. Sliced object
4. Printed it - hadn't noticed the error

### Actual results

Cura Preview shows support extending out past the part of the object that needs supporting. Cura 5.2.1 does not do this.
This screenshot shows the identical object sliced with 5.3 and 5.2.1, with identical settings.
When printed, the support was stuck to the surface of the object.
[image]
[T40sideLH3.5.3mf.zip](https://github.com/Ultimaker/Cura/files/10934224/T40sideLH3.5.3mf.zip)

### Expected results

Support should only be under the "overhanging" part - there is nothing above the extra support to hold up.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

"Prepare" screenshot of object
[image]

MariMakes, 2023-03-17:
Hey @Barry4ad,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I took a look at your project and it seems to be a matter of settings.
If you dial back the Support Horizontal Expansion and turn off Enable Towerr, it looks more like you expected. 

[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

Barry4ad, 2023-03-18:
Hi,

Thanks for your response.

As you suggest, it’s not a bug as such, but…
I opened the project file* that I uploaded, into Cura 5.3. The two settings you suggest changing were not visible in Print Settings.
* I never usually use "save project”.

“Use towers” makes no difference whether checked or unchecked to the protruding support.

However, on making “Support Horizontal Expansion” visible, it comes pre-filled with 0.8mm. Reducing this to 0.0mm, does indeed make the support as I want it to be.
In Cura 5.2.1, this setting seems to be 0.0mm, when first made visible - I think unlikely that I've altered this value, as I’ve only used Support for printing a few times in past years. (Settings and Profiles were imported to 5.3 from 5.2.1 - which came from 4.8).

It seems that a setting containing a value, affects the the slicing, whether it’s visible or hidden - Support Horizontal Expansion certainly does, as I’ve just tried slicing with it set to 0.8mm, visible and hidden, and both slices have the same protruding support.

Perhaps the “Settings visibility” window could have a note warning that a hidden setting containing a value, may not be disabled,  just because it can’t be seen?

Regards,

Barry.

MariMakes, 2023-03-20:
Hey @Barry4ad,

We have a couple of different setting visibility types. You can see the different types by clicking the hamburger next to your search bar. Even if the setting is not available in your setting visibility, you can search for it in the search bar. It will just show up greyer than normal. 
[image]

Regarding "It's unlikely that I've altered this value", we have a team that actively and full-time improves print profile settings. If they believe that all printers can benefit from a changed setting they roll it out to all types of printers. In this case, this https://github.com/Ultimaker/Cura/commit/0b6edce20ab2a8edaebfe62be16aea5c7d537766 changed and introduced the new value. It is mentioned in the release notes like this: 
- Updated Support Horizontal Expansion, Support Join Distance, and Minimum Support Area Settings for UltiMaker printers to improve support reliability.

But looking at the changes made, it seems more like it's applied to all 3D printers instead. It's a bit unfortunate that in your case it doesn't introduce support reliability. I'll bring it up to the team to see if we need to update the release notes to reflect the impact correctly. 

Still, it doesn't seem like a bug but intended behavior. Would you agree I can close the issue?

Barry4ad, 2023-03-26:
> Regarding "It's unlikely that I've altered this value", we have a team that actively and full-time improves print profile settings. If they believe that all printers can benefit from a changed setting they roll it out to all types of printers. In this case, this 0b6edce <https://github.com/Ultimaker/Cura/commit/0b6edce20ab2a8edaebfe62be16aea5c7d537766> changed and introduced the new value. It is mentioned in the release notes like this:

Hi,

Sorry about delay in replying - issue with new computer.

I realise that there are different ways of showing settings, but it isn’t made clear that a hidden setting, will still affect the slicing action. I have suspected this in the past, however.

I have since also noticed that the support horizontal expansion value changes with nozzle size.

I imagine that many people treat Release Notes changes lists like alterations to Credit Card Ts & Cs, and don’t take a lot of notice.

You may close this as a bug, as it seems to be an intended effect.

Long-time niggles: 
Cura still won’t slice an stl file after previously opening a gcode file, without quitting.

Random (90/180 degree) rotation of multiplied model.

Regards,

Barry.

MariMakes, 2023-03-28:
Hey @Barry4ad,

Thanks for the update. 
In that case, I'll close the issue. 

About opening Gcodes. 
In Cura 5.3 I can open a 3mf file after opening a gcode as long as I Clear the Bulidplate before doing so. 
Could that maybe work for you?
[image]

closed 2023-03-28 by MariMakes

Barry4ad, 2023-03-29:
Hi,

Thanks for the suggestion, but... 

Clearing the buildplate and then opening a 3mf file produces the same result as clearing the buildplate and opening an stl file, whether opened as a project, or importing the models. There’s still no option to [Slice] the model, just [Save to Disk], which just saves an empty file, and this admonition:

￼

No slicing unless quit and relaunched. Fortunately, I don’t open gcode files very often, and they are set to usually open in TextEdit.

Cura on Mac has been doing this for years. I think I reported it as a bug on a previous version ages ago.

Regards,

Barry.