Ultimaker/Cura #14831: Skirt line count dont match

PeterSoenberg, 2023-03-09:
### Application Version

5.3.0 alpha xmas

### Platform

Windows 11

### Printer

ender 2

### Reproduction steps

just picture

### Actual results

just picture

### Expected results

just picture

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

PeterSoenberg, 2023-03-09:
Skirt line count dont match the picture

MariMakes, 2023-03-17:
Hey @PeterSoenberg,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I think I know what might be happening. 
Your _Skirt/Brim Minimum Length_ might be too high to have a single skirt layer. 
That setting ensures that at least a certain amount of material is extruded when a skirt is printed.
In smaller models that will result in thicker skirts like this: 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-04-21 by github-actions[bot]