Ultimaker/Cura #14832: Multiple feature stacks parts on top of each other

jrnewman32, 2023-03-09:
### Application Version

5.3.0

### Platform

Ventura 13.2.1

### Printer

AnyCubic Vyper

### Reproduction steps

1. Open .3mf file
2. Centered it on build plate with command r
3. With part selected, pressed command m
4. Selected 1 copy and selected OK.

### Actual results

The new part was placed on top of the original part, but was shifted so it created a completely new part. 

### Expected results

There should be two separate parts placed on the build plate with adequate spacing between them.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

After doing some more investigation, it appears that this only happens when 1 additional copy is added to the build plate. It works fine with 2 or more copies. So a good work around is to add 2 copies and then deleted one.

Here is the image of just original part.
[image]

Here is the image after the 1 copy was added to the build plate:
[image]

Here is the image after 2 copies are added to the build plate:
[image]

MariMakes, 2023-03-17:
Hey @jrnewman32,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Can you check a setting for me in your preferences?
Do you have _Ensure Models Are Kept Apart_ enabled?
That settings should prevent this from happening. 

[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

jrnewman32, 2023-03-17:
Thanks for the reply. Yes, for some reason this “Ensure Models Are Kept Apart” wasn’t selected. I don’t remember having any problems with this issue before with earlier versions or having to set it. Maybe the last update didn’t pickup this initial setting. Not sure why adding more than one model worked. I would have thought it would have stacked them all. Also, I’m not sure why you wouldn’t want to keep multiple models apart in the first place. Seems like it should be set by default. Anyway, glad it was a setup error on my part. Sorry for the bother. This software has always been pretty rock solid. The only problems I’ve seen are crashes when I leave it open and then the Mac goes to sleep. That’s probably going to be a hard thing to track down. Thanks,Jerry Newman

jrnewman32, 2023-03-17:
Sorry for the second reply. But I did some more testing of the multiple models and found another issue with the multiple function with adding a single part. I tried different model that I hadn’t ever made multiple parts at one time.

 ￼

This image shows the part on the build plate after using the command r keys. I then added another parts using command m keys and selecting 1 and pressing ok button. 

￼

The above image shows that it place the new part to the right of the original part. Because of the length of this part, the new parts isn’t all placed on the build plate. So when you slice them, it only slices the part that is completely on the build plate, as you would expect. This is shown in the image below.

￼

The image below shows how the parts were placed using the command r keys after the second part was added to the right of the original part. In other words, the command r keys properly placed the parts so they both can be sliced. 
￼

Just for grins, I tried adding two more parts (total of three) to the build plate. Using the command m keys and selecting 2 and then pressing ok button placed all three parts in the center of the build plate and were sliced, as you can see in the image below. This is how I would expect it to function.

￼

So in conclusion, there still seems be something wrong with the way the multiple addition of a single part is functioning. 

Please review this and let me know.

Thanks again for your time,
Jerry Newman

MariMakes, 2023-03-20:
Hey @jrnewman32,

Thanks for the update and the confirmation 👍 

I really appreciate the follow-up with the different steps, but the pictures got lost on GitHub. 
Do I understand correctly that the parts are not overlapping, but they are not arranged automatically correctly on the buildplate?  Could your issue be similar to this one? https://github.com/Ultimaker/Cura/issues/14112
We've added a ticket to the backlog with the intent to improve this. 

If that's not it and you still want to show the images, you'll need to open Github in your browser to attach any files to your ticket. You cannot add them in an email reply, unfortunately. 
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-04-21 by github-actions[bot]