Ultimaker/Cura #14834: Crash During Slice

dougmccurry, 2023-03-09:
### Application Version

5.5.2

### Platform

Windows 11

### Printer

Ender 3S1

### Reproduction steps

Click Slice!

### Actual results

Program crashes during slice

### Expected results

Slice should complete
[cura.log](https://github.com/Ultimaker/Cura/files/10937019/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/10937023/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Files uploaded in results

maruhe, 2023-03-10:
seems like project-file is missing
not a solution but try to move or rotate your model - if still not working try without support

dougmccurry, 2023-03-10:
I upgraded to the latest Cura last night and it sliced as expected with supports.

Doug McCurry
502-664-2421

MariMakes, 2023-03-17:
Hey @dougmccurry,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
It sounds like that fixed your issue 👍 

I'm glad to hear. I'll close your issue.

closed 2023-03-17 by MariMakes

dougmccurry, 2023-03-17:
Yes sir. All good on 5.3.

Doug McCurry
502-664-2421