Ultimaker/Cura #14838: (5.3.0) multiple objects intersecting after opening

ungutknut, 2023-03-10:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

doesn't matter (most probably)

### Reproduction steps

Open more than 1 .stl file at once or in a row.

### Actual results

The objects won't load with a reasonable distance to each other, but rather intersect.

### Expected results

Cura is supposed to distribute the objects separated over the build plate as it did in previous versions (e.g. tested positive in 5.2.3-alpha-xmas).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

On the screenshot you can see 5.2.3-alpha-xmas on the left- and 5.3.0 on the right side. Both show the buildplate right after loading 3 different objects. In case this isn't reproducible, I can provide a project file.
[image]

MariMakes, 2023-03-17:
Hey @ungutknut,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞

Can you check a setting for me in your preferences?
Do you have Ensure Models Are Kept Apart enabled?
That settings should prevent this from happening.
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore.
Would you still like to keep this issue open or do you agree that we can close it?

MariMakes, 2023-03-20:
Glad to hear!
I'll close this issue

closed 2023-05-09 by rburema

ungutknut, 2023-05-09:
Yes, even with "ensure models are kept apart" ticked on, they frequently managed to intersect.

And there's a similar issue which might be related:
When lifting an object (e.g. move Z +1mm) into the air, so nothing touches the buildplate, cura still makes it touch the buildplate despite the silhouette clearly shows it being lifted. I do this sometimes for almost-flat objects to generate taller supports which are much easier to remove than a 1-layer support and it stopped working with 5.3.0; "automatically drop models to the build plate" is disabled BTW.

If this isn't yet a known issue, I can make a proper report of it and test with 5.3.1.

MariMakes, 2023-05-10:
Hey @ungutknut,

Your report reminds me of [this issue on Github](https://github.com/Ultimaker/Cura/issues/15179)
Can you try the following:

1. Turn off "Generate Support".
2. Go to the Mesh Fixes settings and turn off "Remove Empty First Layers".
3. Turn on "Generate Support"

I believe [it's resolved in the next release with this fix](https://github.com/Ultimaker/CuraEngine/commit/3b1f2e612534f0ed05e022159a5308a6d2c30479).

ungutknut, 2023-05-10:
Thank you! That's the exact same issue and fix.

commit 6a5a9e70 (2023-05-09): Fix whole output of nest2d was translated.

By default, the alignment would cause the entire mass of objects to have its center of gravity to be at the origin, including any fixed object. So there was an unwanted 'whole space' translation that we didn't take into account could occur beforehand. Fortunately, there's an option to just set it not to re-translate the whole space. It's unlcear to me how this would've worked in 5.2.2 and prior (and it did ... somehow), since we didn't change anything, including that default, as far as I can see. Anyway, the arrange function can handle 'fixed' object [... cut]