Ultimaker/Cura #14843: Crash while slicing

iduarte1966, 2023-03-10:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Slice a STL

### Actual results

Can't slice, error.
[image]

### Expected results

No errors
[SpiceGlassHolder_6er.zip](https://github.com/Ultimaker/Cura/files/10946159/SpiceGlassHolder_6er.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10946178/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Not just with this file, many others

MariMakes, 2023-03-17:
Hey @iduarte1966,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

[image]

I was able to slice your file correctly but I see
_Backend exited abnormally with return code 2!_
So I'll have one of the developers take a look to see what that means. 
Fingers crossed 🤞

iduarte1966, 2023-03-17:
Hi
I've reinstalled Cura and is solved. Thank you

closed 2023-03-17 by iduarte1966