Ultimaker/Cura #14847: Print Sequence "One at a Time": Print head returning to build plate outside the print area of any model

Bastel-Bodo, 2023-03-11:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom FFF printer (Original Prusa MINI+)

### Reproduction steps

I tried to print two "S"-shape hook objects in "One at a Time" mode. Print head dimensions were correctly dialled in, and on the "Prepare" screen, the preview polygon showing the print head boundaries during the print showed sufficient clearance from the first printed part during the print of the second one.

[image]

However, this turned out to not be the case after slicing.

### Actual results

In the sliced result, Cura tried to move the print head down in the middle between the two printed parts instead of inside the boundaries of the second part's skirt. This lead to the print head colliding with the first printed part (which, fortunately, I was able to catch before any damage to my printer occurred).

You can see the print head's initial position after the move back to the build plate at the start of the second object's first layer (layer 81 in the attached project) in this screenshot:

[image]
[image]

### Expected results

When the print head is moved back to the build plate after finishing an object in "One at a Time" mode and starting the next one, Cura should ensure that the print head is moved inside the boundaries of the part (or its skirt/brim) on the X/Y axes before moving it down on the Z axis, so that the boundaries the print head actually stays in match the preview on the "Prepare" screen. Otherwise, there is a risk of damage to the printer.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file for reproducing the issue:
[Hook_HeadCrash.3mf.zip](https://github.com/Ultimaker/Cura/files/10948978/Hook_HeadCrash.3mf.zip)

MariMakes, 2023-04-03:
Hey @Bastel-Bodo,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I took a look at your project but I'm having some trouble understanding what your problem is. Are you worried about a potential collision, or have you observed the actual collision of the models? 

When printing one at a time your models have way bigger shadows to prevent potential collisions. If you place your models on the buildplate with the shadows in mind, collision shouldn't occur. 
[image]

If you want you can also have more control over where a layer starts and ends with the following settings:
- Layer Start X
- Layer Start Y
- Z Seam Alignment User Specified. 

Do you have a picture or a video of a collision that occurs? That would help me to bring your issue up to the team and to get a high priority to implement.

Bastel-Bodo, 2023-04-04:
Hi @MariMakes ,
no worries. 😉

So, my problem is that I observed an actual collision of the print head with the first printed part when it moved to the second one.

My print head extends about 38 mm to the left from the tip of the nozzle. According to my understanding (and please correct me if I'm wrong here), the dark shadow shows the area in which the tip of the nozzle is moving (in my case the skirt), and the lighter shadow shows the area that the print head stays in, according to the extents of the print head specified in the printer settings.

Now, according to the light shadow around the second part, the print head should've stayed well clear of the first printed part while printing the second one. But that wasn't the case. From the looks of it, Cura instead took the smallest X and Y coordinates of the skirt of the second part and then moved the nozzle back to the build there (like the lower-left corner of a bounding rectangle around the skirt). But that point was way outside the area of the dark shadow on the second part, and thus the print head extents also were outside what the lighter shadow indicated.

I understand that you can set the layer start XY coordinates. However, the shadows should already show you if you're good with your object placement or not. If Cura moves the print head down outside the indicated areas, then the shadows create a false sense of safety.

MariMakes, 2023-04-06:
Hey @Bastel-Bodo,

Thanks for the update. 

I brought your issue up to the team. 
There is no clearer explanation than the one that can be found int the [Cura Settings Guide which can be found here.](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/blackmagic/print_sequence.md)

> According to my understanding (and please correct me if I'm wrong here), the dark shadow shows the area in which the tip of the nozzle is moving (in my case the skirt), and the lighter shadow shows [...]
That's also my understanding of it. 

They suggested that perhaps you can doublecheck if the Printhead Dimension Settings are correct. If they do not match the actual dimensions of your head, that might have caused the collision. 
[image]

Bastel-Bodo, 2023-04-10:
Hi @MariMakes ,

my print head size is set correctly. I even dialled in a few millimeters more than the actual size, to be safe. 
But the collision still occurred. And it's not like the print head just barely touched the first object: It overlapped with it by about 6 mm when returning to the build plate.

That's why I'm bringing this issue up: Had the print head stayed in the area indicated by the shadows upon returning to the build plate, it should never have come down on top of the first object. But it returned outside the second object's print area, causing the crash.