Ultimaker/Cura #14850: Cura File Slicing Issue

popculturezoo, 2023-03-11:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Creality Ender-6

### Reproduction steps

Imported file 1594-6-_9_body_tail_full_FDM (repaired).stl

Settings were already set per previous successful slices

Clicked Slice

Progress bar gets about 3/4 of the way and then an error pops up "Slicing failed with unexpected error"
[image]

### Actual results

Stops at error. Trying again produces the same results.

### Expected results

I was expecting a successful slice.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The relevant files should all be in the .zip file
[cura error.zip](https://github.com/Ultimaker/Cura/files/10949235/cura.error.zip)

MariMakes, 2023-03-17:
Hey @popculturezoo,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 contains the fix for this for a number of slicing crashes. It seems to be the case for your model aswell. 🎉 
[image]

You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

popculturezoo, 2023-03-17:
It can be closed. I'll download the new Cura. Always happy for a new
version as it just keeps getting better and better!

MariMakes, 2023-03-17:
That's so sweet. Thank you ❤️ 
Apologies for the delay.

Happy printing!

closed 2023-03-17 by MariMakes