Ultimaker/Cura #14854: Sometimes, the supports are created on the air, having space in the floor to start creating on it

Ignacio1224, 2023-03-12:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Artillery X1

### Reproduction steps

1. Import the model
2. Set support mode to "tree"
3. Change the angle of generating supports

### Actual results

Te entire support structure is well, but there are some cases that the support was generated on the air, having space in the floor to start on it.

### Expected results

I expect that all support structure generated was started from a solid body, could be the floor or the support wall or something, this with the option "only touching build plate".

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[AlienError.zip](https://github.com/Ultimaker/Cura/files/10950148/AlienError.zip)
[image]
[image]

MariMakes, 2023-03-21:
Hey @Ignacio1224,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
That's a super impressive and difficult model to print 😮 

Did you know that we are working on improving the Tree Support?
There is an Alpha available if you are interested in experimenting with it. 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-alpha%2Bxmas

It looks pretty promising for your model. 
[image]

I'll include this model in our test suite to see if we can improve it for the next release.

Ignacio1224, 2023-03-22:
Hey @MariMakes 

Nice to have a response, sure, I'm testing it and seems ti be better than the last relese but it has the same bug I found :(

I tried to reduce the supoort angle like 30 degrees or so, and it worked better, however the supports in the air still happening. Another thing I made, is rotate the model, increase the support contact surface to 1mm to ensure that almost a part is not in the air and is more rigid, then I printed it.

It takes a day to print and the quality was succesfully, I did not pay attention to the supports, but, at removing it and looking for the final result of the areas which are supported, were pritty good for me.

This is a kind of soluttion that worked for me, but I had to experiment and overthink hehehe