Ultimaker/Cura #14858: CURA CRASHES AND LAGS

sevin13x, 2023-03-12:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

ankermake M5

### Reproduction steps

1. open cura
2.  anything will cause lag and crashes. A simple mouse click will turn screen white. Ive tried many versions all with the same results. i use a surface pro laptop with AMD ryzen 5 I also use Cura LE (lulzbot) and it runs beautiful unless i add my Ankermake M5 profile to it and then it will open to the dreaded WHITE screen of death! lol PLEASE get this taken care of due to the fact I HATE PRUSA SLICER!

### Actual results

laggggggggg OR crash

### Expected results

Able to slice a model!

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Im pretty good with computers and i cant find these log files ANYWHERE i guess ill keep looking and send when i find trhem

haithamkob, 2023-03-12:
I also have the same problem in all versions 5.3.0 ,5.2.2 after updating plugin Sidebar GUI,, how can i uninstall this plugin.

haithamkob, 2023-03-12:
i just deleted the folder for Sidebar and every thing id working normal.
C:\Users\haith\AppData\Roaming\cura\5.3\plugins
C:\Users\haith\AppData\Roaming\cura\5.2\plugins

sevin13x, 2023-03-12:
Damnnnn ! It looks like that was it! Thank you very much for your help my
friend!

Thank you!
Sevin Hadley

haithamkob, 2023-03-13:
Hope they will solve the problem with Sidebar soon.

fieldOfView, 2023-03-13:
For more information, see https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/29

Unfortunately I currently cannot reproduce the issue. Turns out I can only reproduce it sometimes - in combination with Powertoys FancyZones, which makes fixing it hard.

haithamkob, 2023-03-21:
I have 2 computers one win10 and win11.
Version 4.6 worked well on win10
But win11 did not work keeps lag and crashes.

sevin13x, 2023-03-21:
> Hey all,
>
> Apologies that it took me a while to land here.
> Glad to see that this is resolved.
>
> Do I see correctly that it seems to be a duplicate of #13820
> <https://github.com/Ultimaker/Cura/issues/13820> ?
>
> To me, it doesn't seem like a bug in Ultimaker Cura anymore.
> Would you still like to keep this issue open or do you agree that we can
> close it?

It would lag like a slow motion horror movie! LOL! Deleting sidebar
settings make it work like it should. That other issue you mentioned seemed
a little different. Also I think it has something to do with Windows 11 and
AMD processor

Thank you!
Sevin Hadley

MariMakes, 2023-03-23:
I'll be closing this issue since there is a workaround in place. I hope you understand.

sevin13x, 2023-03-24:
Thank you!

Thank you!
Sevin Hadley