Ultimaker/Cura #14862: Brim distance on multiple object

Pentecost95, 2023-03-12:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ender

### Reproduction steps

1. general brim distance 0
2. adding brim distance 0.5 feature on different object using built in feature "Mesh type : Normal model" and adding istruction to single object that i want

### Actual results

every object has 0 brim distance

### Expected results

object that i selected and adding 0.5 should have that instead remaining 0

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no need additional... just adding feature on single ibject doesn t take effect

MariMakes, 2023-04-03:
Hey @Pentecost95,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I took a look and I can reproduce your bug. 
[image]

We introduced some changes to the brim in Cura 5.3, this bug must have snuck in during development.

I'll bring it up to the team to see what they can do to improve it. 
It's a feature I use myself as well, so fingers crossed that it can be resolved 🤞. 

Thanks for the report! 👍

Pentecost95, 2023-04-08:
> Hey @Pentecost95,
> 
> I brought your issue up with the team, they agree it's worth fixing. We've added a ticket to the backlog with the intent to improve this. For internal reference [CURA-10503](https://ultimaker.atlassian.net/browse/CURA-10503)
> 
> Thanks for the report! 👍

Thank you for considering this! Hope will fix soon bye ;)

Pentecost95, 2023-05-05:
> I'll open this issue again since it's not resolved yet ant this is still something we want to improve.

sorry i' ve closed by mistake! should be fixed soon yes, it s an important feature bug, thanks!

tfjield, 2025-05-08:
Hi, this still is not fixed in 5.10.0...