Ultimaker/Cura #14866: Segfault when installing multiple plugins

BagelOrb, 2023-03-13:
### Application Version

5.2.2

### Platform

Ubuntu 22.04

### Printer

UM3

### Reproduction steps

1. Open marketplace
2. Install Mesh tools
3. accept
4. install settings guide, auto orientation
5. get two accept screens
6. accept the first
7. segfault

### Actual results

segfault

### Expected results

plugins install happily without giving seg the fault

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

cmd output:

`(base) tk@tk-Katana-GF66-12UEOK:~/Downloads$ ./UltiMaker-Cura-5.2.2-linux-modern.AppImage 
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltiMaBq0EN5/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaBq0EN5/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
Gtk-Message: 12:28:51.812: Not loading module "atk-bridge": The functionality is provided by GTK natively. Please try to not load it.
Gtk-Message: 12:31:20.881: Not loading module "atk-bridge": The functionality is provided by GTK natively. Please try to not load it.
qt.qml.context: file:///tmp/.mount_UltiMaBq0EN5/share/cura/resources/qml/WelcomePages/CloudContent.qml:22:5 Parameter "newCloudPrintersDetected" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
Segmentation fault
`

logs:
`2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Application.initialize [163]: Initializing UltiMaker Cura
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Application.initialize [164]: App Version 5.2.2
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Application.initialize [165]: Api Version 8.2.0
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Application.initialize [166]: Build type None
2023-03-13 12:28:41,745 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib/uranium must be a folder.
2023-03-13 12:28:41,745 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib64/uranium must be a folder.
2023-03-13 12:28:41,745 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib32/uranium must be a folder.
2023-03-13 12:28:41,745 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/.mount_UltiMaBq0EN5/plugins must be a folder.
2023-03-13 12:28:41,745 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/Resources/uranium/plugins must be a folder.
2023-03-13 12:28:41,745 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/Resources/cura/plugins must be a folder.
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is /home/tk/.config/cura/5.2
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is /home/tk/.local/share/cura/5.2
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is /home/tk/.cache/cura/5.2
2023-03-13 12:28:41,745 - INFO - [MainThread] UM.Qt.QtApplication.initialize [164]: Preferences file not found, ignore and use default language 'en_US'
2023-03-13 12:28:41,745 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMaBq0EN5/share/cura/resources/bundled_packages/cura.json
2023-03-13 12:28:41,745 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMaBq0EN5/share/uranium/resources/bundled_packages/uranium.json
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.6 Core profile
2023-03-13 12:28:41,745 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile
[... 256 lines cut ...]
2023-03-13 12:34:29,984 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [91]: Updating IntentSelectionModel.
2023-03-13 12:34:29,984 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [91]: Updating IntentSelectionModel.
2023-03-13 12:34:29,985 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [91]: Updating IntentSelectionModel.
2023-03-13 12:34:30,007 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [168]: Need to check for errors again. Discard the current progress and reschedule a check.
2023-03-13 12:34:30,016 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/tmp/.mount_UltiMaBq0EN5/share/uranium/resources/shaders/default.shader]...
2023-03-13 12:34:30,017 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/tmp/.mount_UltiMaBq0EN5/share/cura/resources/shaders/grid.shader]...
2023-03-13 12:34:30,129 - INFO - [JobQueueWorker [2]] FirmwareUpdateChecker.FirmwareUpdateCheckerJob.run [109]: Reading firmware version of Ultimaker 3: checked = 0.0.0 - latest = 5.3.0-20211003.0
2023-03-13 12:34:30,156 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2023-03-13 12:34:30,249 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [164]: New error check scheduled.
2023-03-13 12:34:30,607 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [221]: Error check finished, result = False, time = 0.36s
2023-03-13 12:34:46,662 - WARNING - [MainThread] UM.Qt.Bindings.Theme.getColor [120]: No color primary_text_hover defined in Theme
2023-03-13 12:34:48,585 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.03607034683227539 seconds
2023-03-13 12:34:51,460 - DEBUG - [MainThread] UM.PackageManager.getPackageLicense [696]: Found potential license file '/files/plugins/MeshTools/LICENSE'
2023-03-13 12:34:52,731 - DEBUG - [MainThread] UM.PackageManager.getPackageInfo [668]: Found potential package.json file '/package.json'
2023-03-13 12:34:52,732 - INFO - [MainThread] UM.PackageManager.installPackage [505]: Package [MeshTools] version [3.7.0] is scheduled to be installed.
2023-03-13 12:34:52,732 - INFO - [MainThread] UM.PackageManager._saveManagementData [274]: Package management file /home/tk/.local/share/cura/5.2/packages.json was saved
2023-03-13 12:34:52,733 - DEBUG - [MainThread] Marketplace.CloudSync.CloudApiClient._subscribe [44]: Subscribing to MeshTools
2023-03-13 12:34:53,183 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [369]: request[593de736][put][PyQt6.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=None][b'{"data": {'...] got an QNetworkReplyError Error transferring https://api.ultimaker.com/cura-packages/v1/user/packages - server replied: . The server returned: b'{"errors": [{"id": "0204bd2f-191d-4bd4-b7ac-a9e40048a5cb", "code": "recordNotFound", "http_status": "404", "title": "The given record does not exist.", "meta": {"record_name": "package_build", "translatable": "The given %{record_name} does not exist.", "error_type": "ModelValidationError", "url": "https://api.ultimaker.com/user/packages"}}]}'
2023-03-13 12:35:14,928 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [169]: Synced cloud printers with account.
2023-03-13 12:35:17,080 - DEBUG - [MainThread] UM.PackageManager.getPackageLicense [696]: Found potential license file '/files/plugins/OrientationPlugin/LICENSE'
2023-03-13 12:35:17,113 - DEBUG - [MainThread] UM.PackageManager.getPackageLicense [696]: Found potential license file '/files/plugins/SettingsGuide2/LICENSE.md'
2023-03-13 12:35:18,606 - DEBUG - [MainThread] UM.PackageManager.getPackageInfo [668]: Found potential package.json file '/package.json'
2023-03-13 12:35:18,606 - INFO - [MainThread] UM.PackageManager.installPackage [505]: Package [SettingsGuide2] version [2.9.2] is scheduled to be installed.
2023-03-13 12:35:18,615 - INFO - [MainThread] UM.PackageManager._saveManagementData [274]: Package management file /home/tk/.local/share/cura/5.2/packages.json was saved
2023-03-13 12:35:18,616 - DEBUG - [MainThread] Marketplace.CloudSync.CloudApiClient._subscribe [44]: Subscribing to SettingsGuide2`

MariMakes, 2023-03-14:
Hey @BagelOrb,

Good to see you 😄 
I'm surprised you are not on Cura 5.3 😈 

I have some trouble reproducing your issue in Cura 5.3, I'm able to install multiple plug-ins at once without a segfault. 

I'll have someone from the team take a look at your logs. 
Fingers crossed 🤞

rburema, 2023-03-24:
Hi! I think this is a Linux issue. On Windows, the acceptance screens are modular, there was no way to accept the first window without accepting the second window first. (Just tried this with 5.2.x)

nallath, 2023-03-29:
I've been able to reproduce it, but it's tricky. The screens are modular, so you can't actually click the other one. If you click the one on top it does segfault

nallath, 2023-07-25:
Since it's hard to reproduce and probably even harder to fix, I'm going to close this as won't do.

closed 2023-07-25 by nallath