Ultimaker/Cura #14872: Outer Wall Wipe Distance works in an opposite way.

IDizor, 2023-03-13:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Prusa i3 Mk2

### Reproduction steps

1. 3d model details: Vertical pipe with wall width 0.8mm. Nozzle 0.4mm.
2. Cura settings:
Wall thickness = 0.4;
Outer Wall Wipe Distance = 0.5;
Wall Ordering = Inside to Outside;

### Actual results

Outer Wall Wipe Distance works in an opposite way (moves nozzle backward), and produces horrible Z seam.

### Expected results

Outer Wall Wipe Distance should keep nozzle moving forward.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[TestPipe.zip](https://github.com/Ultimaker/Cura/files/10960922/TestPipe.zip)

PS: Selecting another Wall Ordering will fix this issue, but I need to select 'Inside to Outside' to have another model parts printed correctly.

MariMakes, 2023-03-14:
Hey @IDizor,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I believe I understand what you are running into. 
The nozzle moves a little back before it continues squishing and moving the material. 
[image]

At first I thought it was related to the Arachne Engine but it seems to have been there for a while. 
The whole move is gone If I reduce the Outer Wall Wipe Distance to 0.
Can you  share why the Outer Wall Wipe Distance is so important for your printer/printjob or why you need it to be higher than 0.2?

IDizor, 2023-03-14:
It is simple. I'm using Outer Wall Wipe Distance to minimize Z seam and need nozzle to keep moving forward. And I need to set Wall Ordering = Inside to Outside to get better results for overhangs. Thank you.

IDizor, 2023-03-14:
If I set the Wall Ordering to 'Outside to Inside' the Outer Wall Wipe Distance works correctly. But I need 'Inside to Outside'.

Timocop, 2023-03-15:
Wall ordering seemd to be broken in 5.3

As seen here https://github.com/Ultimaker/Cura/issues/14851#issuecomment-1465827973

MariMakes, 2023-03-28:
Hey @IDizor,

Quick update on our side. 
We had a look and it indeed seems to show the incorrect behavior. 

But we also see that the correct behavior returns when the model isn't just two outside walls. 
When making the model a little thicker and increasing the wall line count you can see that the correct behavior returns. 
[image]

Is that an option for you to get unstuck? 
Are you printing this exact model and is dimensional accuracy important?

IDizor, 2023-04-02:
Thank you @MariMakes 
This is not an option for my case, but I already printed all that I need with Wall Ordering 'Outside to Inside'.

Anyway I think this bug should be tracked and fixed some day.

TheNecronomica, 2023-08-14:
i am also having this issue, the model i am trying to print also ONLY has two walls in the place where i am trying to minimize z seam. Changing inside/outside print order also changes the direction of the wipe, i assume this is not intentional and creates issues whit models where i need the opposite setting for print order compared to wipe direction.