Ultimaker/Cura #14873: Randomly missing part of brim (5.3.0)

Vipcioo, 2023-03-13:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Kossel Plus

### Reproduction steps

1. Multiply selected model.
2. I changed the number of copied items. I also changed their location and amount of brim lines..
The problem only came up with brim for a few/many items.

No message or warning appeared on the screen.

### Actual results

Missing part of brim around models.

### Expected results

Full brim around models.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10960998/cura.log)

5axes, 2023-03-14:
Nice bug ... But if you don't share a sample project the chance that someone can reproduce your issue are nearly null.

Vipcioo, 2023-03-15:
What else I noticed today is that when I move the same model on the bed I get different results.
[image]
[image]
[image]
[image]
[image]

MariMakes, 2023-04-03:
I'll close the issue since it's a duplicate.