Ultimaker/Cura #14874: Material Flow setting interacts badly with First Layer Flow.

see3d, 2023-03-13:
### Application Version

5.1.0

### Platform

MacOS M1

### Printer

Custom: Monoprice Delta Pro Lerdge X Klipper

### Reproduction steps

Sliced a small 2 layer square.  Flow 100%, 1st layer Flow 100%
Generate gcode.

Sliced same part with Flow 120%, 1st layer Flow 150%
Generate gcode.

Compare gcodes and calculate difference in E values on layer 0 and 1

### Actual results

Calculated flow of 1st layer for 150% is 180%  (probably 120% x 150%)
Calculated flow of 2nd layer for 120% is 162%

### Expected results

150% for 1st layer
120% for other layers

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Note: The problem showed up with massive over-extrusion on a large complex part.  The problem did not show up when manually setting the flow with an M221 S150 and M221 S120 for each layer.  The problem also did not show up with an M221 S120 at the beginning of the print with the slicer Flow at 100% and 1st layer flow at 125%.
The steps above were part of the debug process to isolate the problem.

MariMakes, 2023-03-14:
Hey @see3d,

Thanks for your report 👍

This is outside my field of expertise. 
I'll bring it up to the team to see what they can do to improve it.

Are you aware that in Cura 5.2 we introduced different Flow settings? That lets you tune these settings even more precisely to reduce elephants' foot.  
[image]

Can you share why you need to change the flow so extremely?

see3d, 2023-03-14:
Hello MariMakes,  

Yes, I was aware of all the new flow settings.  They are a good addition.

I dial in my machine settings so that I get the initial layer height and line width required for most prints.  

However, I am making some production functional parts in PETG which need to over-extrude or under-extrude to get the aesthetics desired on the bottom layer, or to get the strength in thin wall structures (ribs), to withstand bending without losing layer adhesion.  Over-extruding walls can gain strength without increasing print time on a large part.  The flow can be experimentally determined by adjusting the flow dynamically while a part is printing, then those numbers can be put into the slicer profile for that part.  Adjusting the flow on a part by part basis is simpler and makes more sense than altering my base level machine settings.  I would not go to such measures for one-off parts, but when making many identical parts, it is worth getting the optimum results at the minimum print time.

dmitrym1, 2023-03-19:
As a user I can confirm that I see _**Actual 1st layer flow = (Material flow * First layer flow)**_, and my expectation matches [see3d](https://github.com/see3d)'s : _**Actual 1st layer flow = First layer flow**_

MariMakes, 2023-08-29:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue regarding your the initial layer flow and reminds me of this ticket https://github.com/Ultimaker/Cura/issues/6521. I believe they might be related.

see3d, 2023-08-29:
Thanks for pointing this out.  Yes, this is the same issue as the 6521 issue.  It is a real PITA and should be fixed so that others are not pulling out their hair trying to figure out why their print settings are not responding in a logical way.  It should have been a very easy fix.

MariMakes, 2023-08-29:
🙃 So this confuses our profile experts as well, which makes it easy to convince the team that this would be an improvement. 

This multiplication seems to [have been in there since Cura 4.5.](https://github.com/Ultimaker/CuraEngine/commit/93e1b0baa1960093dcf273b8ae73c56994b74be0)
We'll do a risk analysis of how big the impact of this change is.  
If it's acceptable then we can hopefully change it so it's better understandable. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10983

Thanks for the report! 👍

trankillity, 2023-10-28:
Came here to add another voice to this concern. I just experienced the same today under a slightly different scenario which is very likely related to this.

Initial Layer Height: 0.24mm
Layer Height: 0.2mm
Flow: 100%

The resulting initial layer was barely adhering as it looks like the multiplication for the flow wasn't being multiplied by the difference in layer height. I manually changed the flow mid-print in Fluidd to 120% (the exact difference between the Layer Height and the Initial Layer Height) and the initial layer then printed perfectly.

This can be somewhat avoided temporarily by setting Initial layer Flow to 120% if your normal Flow is 100%, however it looks like from some of the other reports in this issue that there might be multiple sets of multiplication applying here if your normal Flow isn't 100%.

dmitrym1, 2023-10-29:
@trankillity I think you confused "Initial Layer Height" with "Z Offset". The first one is the layer height of the first layer, same as Layer Height is the height of other layers, while Z Offset allows you to adjust head/bed Z relatively to Z sensor. Layer height has nothing to do with relative flow or adhesion. If you have too much gap in the first layer, you cannot compensate for it with any value of Initial Layer Height. Setting 0.24mm Initial Layer Height with 100% flow is not the same as setting 0.2mm Initial Layer Height with 120%. In the first case you'll have just right amount of extrusion, and if you don't, you need to compensate that with Z offset. In the second case you are having 20% more material extruded than needed, you should not really do this

trankillity, 2023-10-29:
@dim1triy I think you're maybe misinterpreting my reply.

If I set all layer heights to 0.2mm with 100% flow, the layers print totally fine with the correct Z-offset. However, if I set normal layers to 0.2mm and Initial Layer Height to 0.24mm (generally recommended to make a taller first layer to flatten any inconsistencies in tramming), then the first layer has lines that don't properly join and present with poor adhesion. Yet if I increase the flow on the initial layer to 120% (keeping in mind 0.24mm is exactly 20% taller than 0.2mm - so 0.24mm needs 20% more filament (i.e. flow) than 0.2mm), the initial layer prints fine.

So working back from the original issue here, it would seem like there's some strange multiplication happening here. In my example, it would seem like the Flow is being calculated/fixed based off the normal layer height, and isn't being adjusted based on differing layer heights. If this is what's happening, I suspect this would also affect the Experimental feature of Adaptive Layer Height (and would explain why I often get over-extrusion using that setting).