Ultimaker/Cura #14875: Failure to slice .stl

edwardpoe, 2023-03-13:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Anycubic Kobra Max

### Reproduction steps

Loaded an .STL from a vendor (one of many)
Clicked on "SLICE" button.  

### Actual results

Slice started, but then stopped with "Slicing failed with an unexpected error.  Please consider reporting a bug on our issue tracker."

### Expected results

I expected file to be sliced.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log.zip](https://github.com/Ultimaker/Cura/files/10961543/cura.log.zip)
[AKMAX_mansion-1-5.zip](https://github.com/Ultimaker/Cura/files/10961545/AKMAX_mansion-1-5.zip)

MariMakes, 2023-03-14:
Hey @edwardpoe,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍
I was able to recreate your crash on my side. 

I did notice that a common workaround was also applicable in your case. 
I rotated the model 90 degrees and moved it away from the center of the buildplate.
That prevents common rounding errors from happening. 
[image]

I hope that prevents you from being stuck, I'll have someone from the team take a look. 
Thanks again 💪

edwardpoe, 2023-03-14:
Hello!

I was able to successfully slice the model after rotating and moving from
the center as you suggested.  Thanks!

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-1Z](https://ultimaker-o7.sentry.io/issues/4863584036/)