Ultimaker/Cura #14876: Missing part of skirt

Vipcioo, 2023-03-13:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Kossel Delta Plus

### Reproduction steps

1. Just sliced model.
2. I checked the same model with the same settings in an older version for confirmation of the error.
The skirt fragment disappears with a random amount of models on the bed, similar to the brim I reported today.

### Actual results

Random fragment missing skirt.

### Expected results

Full skirt around models.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]

MariMakes, 2023-03-14:
Hey @Vipcioo,

Thanks for your report 👍
Could it be that you have the following experimental settings enabled:
Enable Coasting + Wipe Nozzle Between Layers
[image]

To me, it doesn't seem like a bug in Ultimaker Cura, just the result of a setting combination. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-04-06 by MariMakes

grbd, 2023-06-20:
Just to add I've been seeing the same thing myself with parts such as the endplate from the M4 extruder
Note, I have no experimental settings set
https://github.com/VoronDesign/Mobius-Extruder/releases/tag/M4

I'm going to have a try of 5.4.0 beta to see if it's fixed there

[image]