Ultimaker/Cura #14877: Anycubic Kobra Go profile doesn't work with my Go printer, but reverting back to Anycubic Kobra profile works great!

supergoofball, 2023-03-13:
### Application Version

5.3.0

### Platform

Windows 11 / Radeon 7900XTX

### Printer

Anycubic Kobra Go

### Reproduction steps

I have an Anycubic Kobra Go but the default profile for this printer produces spaghetti. Switching back to the original Anycubic Kobra printer profile works perfectly.

### Actual results

[image]

### Expected results

see above

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Very new to 3d printing so not really sure how useful I can be in trouble shooting. Basically, just had to go back to the Kobra profile (instead of the Kobra Go profile) and things went back to normal.

Asterchades, 2023-03-16:
> I can see that the Kobra Go has been contributed at the end of December, by a community member.

Just for my own information, what would be the process for reporting issues with such a submission? This definition has clearly not been constructed properly - each of its material profiles includes redundant comparisons (comparing to True or False directly will never change the result), and the Start GCode not only uses incorrect placeholder values (they appear to be for a Slic3r derivative, most probably PrusaSlicer) but goes on to break them by inserting line feeds in the middle of them.

MariMakes, 2023-04-06:
Hey @Asterchades,

I double-checked what the procedure is for printer definitions. 
I've been told that it's like a wall of graffiti, a bunch of improvements on top of each other. 
You or anyone else is more than welcome to create a pull request to improve the definition.

I can also help with that if you are stuck.

closed 2023-04-07 by github-actions[bot]