Ultimaker/Cura #14878: Malyan MA 10 mini?!?!

fellscreen, 2023-03-13:
### Application Version

5.2.2

### Platform

windows 10

### Printer

Malyan M180 and Malyan M200

### Reproduction steps

1. during printing none vase mode, my printer Malyan MA10 mini kept on moving the extruder but didn't let any filament come out for few seconds and then carry on printing at the NEW extruder precision. that was with the standard 0.4 mm nozzle. other times on same print near the start and at the end the print failed.
2. in vase mode with 1 mm nozzle, not a 0.1 mm but an actualy 1 mm nozzle in vase mode, and the print wasnt water tight at base of the print and failed at end of a completed print too.
3. also in vase mode i tried to print on raft, it too failed. either by taking base off with the raft because somehow it fused to the raft. 
4. or parts of the raft broke off as i was trying to pull base off and the raft broke apart some bits stuck to the print.
5. i tried adjusting the print speed and tried to lower the filaments temp and neither did any of them work too.
6. here is few pictures showing how my completed 3D prints failed.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
the thats getting me the most of there was almost 2000 people said they like and manage to get successful completed print so why can't I. i ev3en tried the second best design, of the same item but slightly different design same results.

### Actual results

i ended up wasting almost a full 1kg reel of PLA+ filament trying to make 1 design work for personal home use in bathroom.

### Expected results

i would like a successful and completed print, and my printer added to the printer profiles please, specially since you got 2 other Malyan products in the list so why not mine too. i got it from amazon for 150 pounds. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

could really do with some help solving this asap, im terminal ill and not sure how long i got left so would like to try finish my futures plans with my 3d printer. but if i cant get this fixed asap then may as well just stop now and give up.

[stderr.log](https://github.com/Ultimaker/Cura/files/10962709/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/10962710/stdout.log)
[0.4 mm pla+ none spiral.zip](https://github.com/Ultimaker/Cura/files/10962781/0.4.mm.pla%2B.none.spiral.zip)

fellscreen, 2023-03-13:
PS i also sent same bug reports to the manufactures of my printer too in china since where they base when i bought my printer from amazon.co.uk and still waiting from them too see if its a fail with printer and not cura to cover all my bases.

MariMakes, 2023-03-14:
Hey @fellscreen,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍
Sorry to hear you are having such a frustrating experience. 😞 

I believe it's an issue that can be fixed if we find you the correct settings for your print job. 
Printing a vase on a raft for example is not so common, and I'm seeing a lot of warping which is usually an indication of incorrect temperature settings. 
Getting your vase watertight, is always challenging for any 3D printer expert and needs a lot of tuning. 

Perhaps it's better to start tuning the specific temperature and fan settings on your printer. 
You can download a plug-in with calibration shapes you can use to experiment with. 
You can get the plug-in here: https://marketplace.ultimaker.com/app/cura/plugins/5axes/CalibrationShapes

To me, it doesn't seem like a bug in Ultimaker Cura.
If you want to start a discussion on the best settings for your printjob, [our UltiMaker community ](https://community.ultimaker.com/) is a better place to do it. You will get more helpful responses if you include a project file instead of a .curaprofile. It contains the printer and settings we need for troubleshooting.  To save a project file go to File -> Save project. 

Would you still like to keep this issue open or do you agree that we can close it?

fellscreen, 2023-03-14:
what about the printer profile as my 3D printer is not on your list of
profiles. you got Malyan m180 and m200 but there no Malyan MA10 mini could
that be fix/or added please so im not picking a different printer profile
thats doesnt belong to mine?

about the prints, so you dont have any idea what might be causing the
problem then? if its such a common problem then how come it hasnt been fix
yet?

MariMakes, 2023-03-14:
Hey @fellscreen,

All non-Ultimaker printers have been contributed by our lovely community members. 
Since Cura is open source, you or anyone else is more than welcome to work on adding this 3d printer and create a pull request yourself. You can find instructions on how to add a new 3D printer definition here: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura 
If you are stuck in any way, please reach out! I'll be happy to help out. 

It can have a lot of different causes which are easier to troubleshoot if you include a project file. 
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

fellscreen, 2023-03-14:
hello
no sorry i may know few things here and there about PC, but i only got this
3D printer so im beginner. with my illness its getting harder to type with
keyboard i dont have voice active software to help either, it take me to
long to try figger it all out with my dyslexia and illness. so is there
anyway you could help me please.

MariMakes, 2023-03-17:
Hey @fellscreen,

Let me introduce myself, I'm Mari. 
I'm part of a company that makes 3D printers and free 3D printing software called UltiMaker. 
UltiMaker Cura is free open-source software that is built by a team of developers and the people in the community. 

This website is Github, the place where we write the code and collect bug reports and feature requests. 
This is not the best place to learn 3D printing, this is where we make 3D software better. 

There are some very cool places on the internet to find people to help you 3D print like:
- https://www.reddit.com/r/3Dprinting/
- https://discord.com/invite/huvWHBq
- https://www.youtube.com/watch?v=YPAXeBuq9qU
- https://community.ultimaker.com/forum/108-ultimaker-cura/
- https://www.facebook.com/groups/1959877270949115

If you find a bug in Cura, we would love to hear it. 
Good luck on your 3D printing journey!