Ultimaker/Cura #14879: Part of the layer was skipped on the raft

homerender, 2023-03-13:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Anycubic i3 Mega Pro

### Reproduction steps

Open STL file
Click Slice
Preview 3rd layer

### Actual results

big part of 3rd layer is missed

### Expected results

You will see a problem... A missed layer part.
[image]
[Card_Game_Keeper.zip](https://github.com/Ultimaker/Cura/files/10962863/Card_Game_Keeper.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

--

MariMakes, 2023-03-14:
Hey @homerender,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12276
We have a ticket on our backlog with the intent to improve this behavior.
There is a workaround with moving the printjob away from the center of the buildplate. 
This location seemed to work for your printjob
[image]

Apologies for the inconvenience. 
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12276

closed 2023-03-14 by MariMakes

homerender, 2023-03-14:
Ohk, good to know, thank you!