Ultimaker/Cura #14882: UFP filetype

CaryBab, 2023-03-14:
### Application Version

5.3

### Platform

Linux Mint

### Printer

FLsun SR

### Reproduction steps

1. Change filetype to "Ultimaker Format Package"
2. I have to manually rename the filetype to "ufp"
3. Save

### Actual results

The file IS actually saved as UFP format, but the file type does not get changed from "gcode" to "ufp"

### Expected results

The filetype should automatically be changed to "ufp"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This worked on version 4.13.1

MariMakes, 2023-03-14:
Hey @CaryBab,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12659
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12659

closed 2023-03-14 by MariMakes

CaryBab, 2023-03-14:
No, that is not the same problem.  I now get the filename all filled out
like it should be, which is what 12659 is referring to.  But If I try to
change the FILETYPE to UFP, it actually saves a UFP formatted file, but
does not change the FILETYPE to UFP like it should.

CaryBab, 2023-03-14:
https://aur.archlinux.org/packages/cura-bin#comment-895214 has a work around for #12659.  Which is what I'm currently using on my Linux Mint 21.1 distribution.

drohhyn, 2023-07-07:
> No, that is not the same problem. I now get the filename all filled out like it should be, which is what 12659 is referring to. But If I try to change the FILETYPE to UFP, it actually saves a UFP formatted file, but does not change the FILETYPE to UFP like it should.

Same problem here, but suddenly it got worse with version 5.4.0 (not alpha): https://github.com/Ultimaker/Cura/issues/12266#issuecomment-1625161942
now the change does not help anymore, my files are always saved as gcode.