Ultimaker/Cura #14889: Cura segfaults when slicing a specific model and profile

Hashfastr, 2023-03-15:
### Application Version

5.2.2

### Platform

Fedora 37 Linux Flatpak

### Printer

Creality Ender-3

### Reproduction steps

Unsure what exact setting caused this but Cura started core dumping on this STL file. I was tweaking setting using a stringing test file and when I went back to actual printing it started segfaulting. Doesn't happen on other STL files with this profile, and this STL file has no problems with other profiles such as the default Ender-3 dynamic quality profile.

1. Load attached project
2. Attempt to slice the model using said project

### Actual results

Sometimes it hangs, but most of the time, if not all the time, it will just segfault. The slicing fails as if I hit the cancel button and segfault nastiness is spit out to logs (attached).

### Expected results

The STL is sliced into gcode.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura-segfault.log](https://github.com/Ultimaker/Cura/files/10976757/cura-segfault.log)
[cura-segfault.zip](https://github.com/Ultimaker/Cura/files/10976790/cura-segfault.zip)

Hashfastr, 2023-03-15:
Upon further testing, it appears that a maximum resolution equal to or less than 10 times the maximum deviation will cause this, e.g.
```
Crash:
  Max Res: 0.25 or 0.24
  Max Dev: 0.025
Fine:
  Max Res: 0.26
  Max Dev: 0.025
```

MariMakes, 2023-03-17:
Hey @Hashfastr,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
It seems to be the case for this model aswell:
[image]
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-04-21 by github-actions[bot]