Ultimaker/Cura #14892: cant slice

chaimshiffy1, 2023-03-15:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ultimate S5

### Reproduction steps

1. open file.
2. click slice.

### Actual results

error says cant slice, report bug

### Expected results

being sliced and ready for print

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_B1_BASE_MATERIAL_Light.zip](https://github.com/Ultimaker/Cura/files/10981601/UMS5_B1_BASE_MATERIAL_Light.zip)
[image]

MariMakes, 2023-03-21:
Hey @chaimshiffy1,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
Wow that's an amazing model 😍 

I can reproduce your error on my side. 
I'll ask the team to take a look, hopefully they can see what is going wrong.

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-22](https://ultimaker-o7.sentry.io/issues/4864445917/)