Ultimaker/Cura #14896: 5.3.0 Slicing Crash

DazedbyNight, 2023-03-16:
### Application Version

5.3.0

### Platform

Windows 11 Home 10.0.22621 Build 22621

### Printer

Ender 3 Pro, Ender 3 Max Neo

### Reproduction steps

Added .3mf file to build plate, selected profile, Clicked slice. "Unexpected error when slicing" message. STL file and different slicer profiles get same results

### Actual results

Unexpected error message when slicing. 

### Expected results

Expecting Gcode file

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10985503/cura.log)
[CE3PRO moon-globe-mk2.zip](https://github.com/Ultimaker/Cura/files/10985504/CE3PRO.moon-globe-mk2.zip)

DazedbyNight, 2023-03-16:
Took a look at a few other slicing reports. Tried a suggested workaround: Rotated 90 degrees and moved off center X -13.0708, Y -8.5493, slice completed

DazedbyNight, 2023-03-16:
Sorry, closed by mistake.

MariMakes, 2023-04-03:
Hey @DazedbyNight,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
I'm glad you at least found a workaround. 

I can reproduce your crash on my side on Cura 5.3.
I'll have someone from the team take a look 👍 

[image]

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-1N](https://ultimaker-o7.sentry.io/issues/4855789320/)