Ultimaker/Cura #14901: Cura 5 won't launch

lin-ycv, 2023-03-16:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

None

### Reproduction steps

I'm on a fresh install of windows 11, installed Cura, and try to launch it.

### Actual results

it won't launch, no UI pop-up, no splash screen.
I see the process in task manager, but after a few seconds it disappears, no error message, no dialog.
I've tried installing different versions, and they were all the same.

How can I debug this issue?

### Expected results

launches normally.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

a log isn't generated in `%APPDATA%\cura\<Cura version>\` just a plugins folder that's empty

MariMakes, 2023-03-21:
Hey @lin-ycv,

Thanks for your report 👍
Sorry to hear you are stuck😞 

We've seen this report a couple of times for users with multiple monitors. 

I have a couple of questions to narrow it down.
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have two monitors placed above each other? Can you try this workaround from the community? https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105
	3. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	4. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	5. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

If none of these work for you, please let me know and we'll take another look 👍

lin-ycv, 2023-03-22:
@MariMakes 
1,2. I don't have multiple monitors, just the laptop screen
3. Haven't installed any GPU tweaking programs
4. The laptop is using Nvidia Quadro P4000 and Intel HD Graphics 630
5. Don't have Nahimic installed

Interestingly, based on 4, I tried setting cura to a different GPU, it wouldn't run on intel, but launched fine after setting it to nvidia. Thanks.

I'm closing the issue since it's solved, but if you'd like any logs or anything for debugging purposes, please let me know and I'll provide them.

closed 2023-03-22 by lin-ycv

lin-ycv, 2023-04-14:
for those that might have the same issue, another thing to note is to update graphics driver from Nvidia/AMD

Had to reinstall one of my laptops, and reencountered this issue, except after setting cura to use the dedicated GPU, the UI would flash
[image]

After updating from the lastest Microsoft default driver to the lastest Nvidia driver, the issues were solved.

kakooloukia, 2024-03-02:
Thank you, I faced the exact same problem and fix with nvidia tool:
[image]