Ultimaker/Cura #14903: failed slice

critchy1, 2023-03-16:
### Application Version

5.3.0-beta.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

loaded my object to print 
fixed model normals in mesh tools menu
changed number of walls to 3 
changed bed adhesion to none
asked for support to be generatded
set infill pattern to cubic

### Actual results

tells me slicing has experienced an unexpected error and to consider reporting a bug

### Expected results

previous to beta release object sliced as normal

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10989160/cura.log)
[astronaut slicing error project.zip](https://github.com/Ultimaker/Cura/files/10989193/astronaut.slicing.error.project.zip)

critchy1, 2023-03-16:
sorry forgot to add this screen shot

[image]

MariMakes, 2023-03-21:
Hey @critchy1

Thanks for your complete report! 👍

When I loaded your model I noticed this error: 
[image]

That's often an indication that the mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.3 Stable.
[image]

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

closed 2023-04-21 by github-actions[bot]