Ultimaker/Cura #14904: Artillery hornet profile import

lordwolf88, 2023-03-16:
### Application Version

5.3

### Platform

Windows 11

### Printer

Artillery Hronet

### Reproduction steps

Added new printer
choose artillery hornet
trying to add profile from arttilery

### Actual results

get error: not available

### Expected results

adding profile.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[Hornet config.zip](https://github.com/Ultimaker/Cura/files/10989895/Hornet.config.zip)

MariMakes, 2023-04-03:
Hey @lordwolf88,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The profiles you shared are made to be loaded in a Custom FFF printer in Cura. I suspect that they were made for Cura versions before Cura 5.3 because in those versions the Hornet was not available as a printer to be selected. 

[image]

I expect that you don't need the profiles to get good print results for your printer. If you want to be sure, profiles are just text files so you can open and read them to doublecheck the settings. 
[image]

If you need a guide in changing the setting names to the readable settings, you can find all the settings by both of their names here: https://github.com/Ultimaker/Cura/blob/main/resources/definitions/fdmprinter.def.json

I'll close this issue since Cura is showing expected behavior. 

Hope this helps 😊

closed 2023-04-03 by MariMakes

lordwolf88, 2023-04-04:
yes, it is good for the costum printer, but I was glad that in 5.3 the Hornet is already included and the silhouette of the machine is visible, which helps a lot in the placement of the object. Can it be somehow solved so that it can be imported to it as well? Also, I have an Ender-3 S1 pro machine with KLIPPER, and I cannot load the speed profiles downloaded from the creality site due to a similar error.

MariMakes, 2023-04-05:
Hey @lordwolf88,

You are not the first to request this and we want to improve the importing behavior. 
You can follow the progress here: https://github.com/Ultimaker/Cura/issues/14165

Thanks for the report! 👍