Ultimaker/Cura #14909: Bird feeder

iandavys, 2023-03-16:
### Application Version

5.3.0

### Platform

windows 11

### Printer

Flsun Q5

### Reproduction steps

just pressed slice then got an error.

### Actual results

Just get a red cross saying error, please report on git hub!
[FQ5_bird feeder v6 v1.zip](https://github.com/Ultimaker/Cura/files/10993787/FQ5_bird.feeder.v6.v1.zip)

### Expected results

get a Gcode and a time to print out.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Hope you can help.

MariMakes, 2023-04-03:
Hey @iandavys,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

Wow! That's a lot of changed settings. 😮 
When I revert your changes it seems to slice normally, so I believe it's related to one of these changed settings. 
[image]

After some digging, it seems to be related to the Adaptive Layerheigth Setting. 
If you turn that one off, you can slice your printjob normally.

[image]

Hope this helps you get unstuck. 👍 
To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it? 

Happy printing and happy Bird Feeding 😄

iandavys, 2023-04-07:
Wow! Thank you!!Yep, as you stated adapted layer hight!Brilliant!!Yes please close it down!But again great job. Just wish I could afford a better 3D printer.This will keep the birds happy!Take care and thank you!From Ian davys
    	Sent from Outlook for iOS

closed 2023-05-03 by MariMakes