Ultimaker/Cura #14912: Cura ignoring initial layer speed setting unless I choose skirt

rpersak, 2023-03-16:
### Application Version

5.3

### Platform

Windows 11

### Printer

Ender 3 pro

### Reproduction steps

Loaded model and sliced it without Build Plate Adhesion or with Raft

### Actual results

Prints raft and all layers with print speed setting (in my case 200mm/s)

### Expected results

Print initial layers including raft with lower speed set in config (in my case 20mm/s)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10994917/cura.log)

MariMakes, 2023-04-03:
Hey @rpersak,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Did you try changing any of the following settings:
- Number of Slower Layers
- Raft Print Speed
- Raft Base Print Speed
- Initial Layer Print Speed

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

rpersak, 2023-04-06:
Hello,

i don't know how to explain this, but I cannot replicate it anymore. I never changed any of settings, you mention, before printing, I double checked those settings.

If and when I bump into same matter, I will provide you with project file.

BR

closed 2023-04-06 by rpersak