Ultimaker/Cura #14914: Creality Ender 3 s1 pro nozzle size problem

Terrybi, 2023-03-16:
### Application Version

5.3.0

### Platform

Win 11

### Printer

Creality Ender 3 s1 Pro

### Reproduction steps

1. went to settings
2. select extruder 1
3. select nozzle size

### Actual results

the only nozzle size is 0.4 mm nozzle. no other choices

### Expected results

should have multiple nozzle sizes available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I checked the machine profile for the Creality Ender 3 Pro and there are multiple nozzle sizes available. Since the S1 pro is just new out, maybe it's just not included?  Thanks for all the great wok you are doing in Cura. Hope I explained this right.

64bittuning, 2023-03-27:
The Ender3 S1 Pro only has profiles built for the .4 Nozzle. TBH, the "Profiles" for the other machines really arent anything special to begin with. You can easily fix this on your end by modifying a few files, or wait two years for the Cura team to bother pulling in a pull request that will fix it.

Terrybi, 2023-03-27:
Maybe so, but I have confidence in the team. I'm sure they'll get to it. I'm not overly knowledgeable in modifying files in Cura so rather than mess something up, I'd rather keep my fingers out.

Terrybi, 2023-04-03:
No problem with the delay, we all get busy. Well I guess I learned something new, I didn't know how you got all the info you used in development, good to know. I got some ideas from some other people in how to create definitions so I might just give it a try. (update) Actually, one of the team members responded to someone else who reported the same thing and supplied a solution so I tried their solution and it now shows all the nozzles.

Thanks

closed 2023-04-03 by Terrybi