Ultimaker/Cura #14920: Slicing fail

ChezJohnson, 2023-03-16:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

Loaded model, repositioned and commenced slicing

### Actual results

Slicing failed approx 50% complete :(
[CE3PRO_half shell.zip](https://github.com/Ultimaker/Cura/files/10996351/CE3PRO_half.shell.zip)

### Expected results

slicing failed!

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

slicing failed!

MariMakes, 2023-04-03:
Hey @GavSeil,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I can reproduce your crash. So I'll bring it up to the team.
Thanks for your report 👍

I also noticed that if I rotate the model 90 degrees, I can slice it successfully. 
I hope that helps you to get unstuck. 
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker