Ultimaker/Cura #14921: Disable Arachne

tacobotics, 2023-03-17:
### Is your feature request related to a problem?

So I've run into a few people who like myself have gaps in their skin walls due to arachne, and using the exact same profile on an older version of cura fixes the issue. Currently there is no way to disable arachne, leading us to downgrade to cura 4 or use a different slicer such as prusa that allows the option. Using cura 4 means I have to use an older version lacking features I do appreciate, such as the upgraded tree supports, but using prusa means I lose all my profiles and some options I like.

### Describe the solution you'd like

Pretty simple, do something similar to prusa where they have an option for arachne or regular slicing.

### Describe alternatives you've considered

I considered switching to prusa/ss but decided not to due to having to start my profiles from scratch, so instead I will be using cura 4.

### Affected users and/or printers

It'll affect anyone who has been having issues with arachne

benkress, 2023-06-21:
I agree that implementing a way to easily disable the effects of Arachne is necessary.  For now, my solution has been to change a few settings that mitigate some of the issues Arachne is causing in my prints.  

Minimum Wall line Width: set to the same as your nozzle to prevent Arachne from making thinner lines.  It still seems to be able to go wider/thicker, however.
Flow Equalization: I set this to 0 as I was getting inconsistent walls using linear advance and having this setting set to 100%.
Print Thin Walls:  Should be unchecked with Arachne but play with its settings in order to get a little more control.
Use Adaptive Layers:  Linear advance and Arachne were causing issues with this enabled, so I disabled it and am getting better/cleaner results on prints that do not require that feature.  

Arachne also negatively affects the z-seam...  Have been forced to manually place it on prints due to the seam not aligning properly ever since the introduction of Arachne.  There were many hours wasted tuning for the issues that Arachne brought.  We definitely need an easy way to disable/mitigate the effects until the kinks are worked out.

sargonphin, 2024-12-29:
Any updates on this?